commaai/openpilot #28816: nav: zooming improvements

sshane, 2023-07-06:
works on PC for me. @deanlee if you want to make some easy improvements, it looks like it continues to zoom when you pan manually, it probably should stop (like it does if you manually zoom). Can also reset the filter when it resets the view/re-centers

_Originally posted by @sshane in https://github.com/commaai/openpilot/issues/28796#issuecomment-1622629543_

varshneydevansh, 2023-08-18:
Hi @sshane can I work on this as I also have experience with Qt and I looked at the comment https://github.com/commaai/openpilot/pull/28796#issuecomment-1622629543

sshane, 2023-08-18:
sure, any issue open here you feel you could tackle, we're happy to let you

jakethesnake420, 2023-08-19:
https://github.com/commaai/openpilot/pull/29488

jakethesnake420, 2023-08-19:
Unless I'm not understanding the idea, I don't think the filter should be reset when you recenter the view.

varshneydevansh, 2023-08-22:
I will do this in a few days. =)

jakethesnake420, 2023-08-22:
> I will do this in a few days. =)

I think I have completed them. Just waiting for it to be merged

closed 2023-08-24 by sshane