commaai/openpilot #28832: "waiting for GPS" until first vehicle motion

jyoung8607, 2023-07-07:
### Describe the bug

Sometimes on startup, my device remains in GPS SEARCH / nav Waiting for GPS indefinitely, until the vehicle starts moving. Have noticed this three times now, when setting a nav-on-openpilot destination at my starting point.

In the background, it looks like I'm getting a fast GPS fix, but neither `liveLocationKalman.positionGeodetic.valid` nor `liveLocationKalman.gpsOK` flip to True until the vehicle moves.

It doesn't always happen. I think it might only be happening on cold boot, or branch switch/update.

### Provide a route where the issue occurs

c5cbf3352b0f03bb|2023-07-06--12-50-49

### openpilot version

219590885fb7463f39bacbbf1d5274b7251d9e7f

### Additional info

[image]

adeebshihadeh, 2023-07-09:
@nuwandavek is this expected behavior? can we do something about it?

nuwandavek, 2023-07-13:
The check that is being triggered here is https://github.com/commaai/openpilot/blob/8d13ddb943de02a1d8f9f593896aa24ae38db34c/selfdrive/locationd/locationd.cc#L307C18-L307C18. In the `gpsLocation` message ( `qcom` gps), the `bearingAccuracydeg` was 0 and only began to change when the vehicle moved. Since the accuracy default values are 0, we wait for them to be non-zero to assume the gps is on. The final solution here seems to be to use the magnetometer for bearing (and to remove this check in locationd).
@haraschax should we patch this in the meantime?
[image]

haraschax, 2023-07-14:
I think this is a failure of rawgpsd, it should not output 0 for bearing accuracy, I will look into fixing it.

closed 2023-07-18 by adeebshihadeh