commaai/openpilot #28838: no recalibration or alert after mounting change

jyoung8607, 2023-07-07:
### Describe the bug

After a recent change in my device mounting, openpilot should have automatically noticed a change and gone into soft disable while recalibrating. Instead, it remained engageable, and both UI path visualization and actual driving behavior were wrong during two decently long drives.

Calibration info in the UI settings menu continued to show pitch and yaw angles that were definitely not correct for its new position, and probably left over from its old position. It's possible the new temporary mounting position was outside acceptable parameters, but I would expect openpilot to throw `calibrationInvalid` and reject engagement.

### Provide a route where the issue occurs

9ee0b6b5f0922277|2023-07-07--07-57-35

### openpilot version

475a4a23a2bdf1c8cacececc959134fb4669d6ee

YassineYousfi, 2023-07-08:
The new mounting position was indeed outside of the range. 

The main issue here is that the previous mounting pitch and new pitch are very far, which confuses the model, and we rely on the model to trigger recalibration.

Looking into improving the reliability of this.

YassineYousfi, 2023-08-07:
converting to a conversation because this is model related and doesn't track like a traditional issue.