commaai/openpilot #28839: Persistent warning lights and Eyesight vehicle icon at start-up.

RCoaster, 2023-07-07:
### Describe the bug

When starting the car after updating Openpilot to 0.9.3 or after a reboot, the traction control, brake, and ABS warning lights come on and stay on. I have to shut down the car and then restart to make the lights go out and stay out. Also, the Eyesight vehicle icon that appears when turning on the cruise control appears, even with cruise control off. 

When using 0.9.1, those issues were not present (the traction control, brake, and ABS lights would come on, then go out after about a second or 2 after the initial start-up, and the Eyesight vehicle icon would not appear until cruise control is turned on. 

I did not use Openpilot when 0.9.2 was released, so I cannot confirm if the issue happened in 0.9.2. 

### Which car does this affect?

Subaru Ascent Limited 2019

### Provide a route where the issue occurs

1cbe6c024ca751ba|2023-07-07--14-31-33

### openpilot version

0.9.3

sshane, 2023-07-07:
does this happen when you turn the ignition on, or only as soon as the comma three displays the startup text (telling you to pay attention)?

sshane, 2023-07-08:
@RCoaster are you on Discord? We can try a few things to narrow down the issue

RCoaster, 2023-07-08:
It happens as soon as the start up text appears (the display starts showing the front camera along with the path planning). Yes, I'm on Discord.

sshane, 2023-07-11:
Messaged you on Discord, lmk if you're free to try a branch!

sshane, 2023-07-31:
Seems like another brand's query is causing these alerts. @jnewb1 when you can, can you remove a FW version from the Outback on a branch so it runs all the queries to see if this is a problem on ours?

RCoaster, 2023-08-01:
An update: I added the lines that sshane had in the branch he sent me to my own branch based off the release 3 staging of 0.9.4. That took care of the Eyesight icon showing up when starting the vehicle. I was looking through the firmware list and comparing those to what my car was reporting from the logs. I found 1 firmware version for the transmission (b'\x00\xfe\xf9\x00\x00') that my car was reporting but was missing from OP. Once I added the firmware version to the list for my Ascent, the faults stopped. I no longer got the warning lights after a reboot. However, a new problem popped up: when I went to engage OP, I got an error message, saying "OpenPilot Not Available, Gear Selector not in D". The cruise control will engage momentarily (I get a speed set ), then disengage. I've uploaded the logs, and the cruise control engagement happens somewhere between the 7:48:15 to 7:49:00 mark. Here's the route:  1cbe6c024ca751ba|2023-07-31--07-42-29--6 . 

Here's what sshane had to say: Ah I think I may know what's going on. Your car was fuzzy fingerprinting before, meaning it goes through all of the requests from all brands. It's very well possible a FW request from another brand is putting the EyeSight ECU into a bad state, or another ECU
When you added your transmission FP, it stopped querying other brands' requests.

sshane, 2023-08-08:
@jnewb1 does running all the queries cause a fault on our Outback?

jnewb1, 2023-08-08:
c3d1ccb52f5f9d65|2023-08-07--23-50-32

no faults after removing all the fw versions from the outback

sshane, 2023-08-15:
I would do a binary search to narrow down which brand is causing this on the user's car, then you can probably guess which request is causing this

RCoaster, 2024-02-28:
Any updates on this? The Eyesight vehicle icon is still showing up on subsequent versions even with the cruise off. As for the warning lights due to fingerprinting, I've put in a forced fingerprint for my car since. I believe it was caused by a OBD port splitter I was using to accomodate an accelerometer provided by my car insurance company to get a discount on my insurance. I've since completed the program and removed the accelerometer dongle, but kept the forced fingerprint in place since it was more convenient. Also, I'm no longer on Discord.

RCoaster, 2024-03-13:
Gone stale. No response from last comment.

closed 2024-03-13 by RCoaster