commaai/openpilot #28840: Minimize git repo size

adeebshihadeh, 2023-07-07:
openpilot is currently a 2.7GB git clone

ajRiverav, 2023-07-23:
consider git lfs.

adeebshihadeh, 2023-07-23:
We've been good about it and other best practices for the last couple years, but there's lot of objects in the history from before then. Also our release branches (devel, release3, etc.) don't use LFS. This issue is for rewriting the history and cleaning up the bloat from releases.

andiradulescu, 2024-07-10:
This issue is now completed with https://github.com/commaai/openpilot/pull/31562 merged and with https://github.com/commaai/openpilot/pull/32955 soon to be merged.

A couple of things to note:
- the script doesn't touch branches unrelated to master's history (master-ci, nightly, devel, release3, release3-staging, dashcam3, release2) - so objects in these branches don't get LFS-ied
- branches related to master's history get rewritten with master's history
- if the branch fails automatic rewrite, it gets deleted, so the one responsible for the branch can fix the conflicts manually

Keeping this in mind, I did some clone tests with `--single-branch` on both repos, cloning just one branch to check the size:

| Branch         | Original Size   | Rewritten Size |
| -------------- | --------------- | -------------- |
| master         | 419.81 MB + LFS* | 60.13 MB + LFS* |
| master-ci      | 331.80 MB       | 331.80 MB      |
| nightly        | 157.96 MB       | 157.96 MB      |
| devel          | 286.07 MB       | 286.07 MB      |
| release3       | 169.18 MB       | 169.18 MB      |
| dashcam3       | 142.35 MiB      | 142.35 MiB     |
| release2       | 1.61 MiB + LFS  | 1.61 MiB + LFS |
| total (*summed*) | 1508.78 MB      | 1149.10 MB     |
| total (*full clone*)     | 2500 MB         | 1700 MB        |

\* LFS for master is 203.97 MB, same as before

The final output repo, I tested with, can be found here: https://github.com/andi-radulescu-contributor/openpilot

closed 2024-09-04 by maxime-desroches