commaai/openpilot #28857: memory leak when loggerd/encoderd get out of sync

adeebshihadeh, 2023-07-10:


adeebshihadeh, 2023-07-11:
@deanlee want to look into this?

deanlee, 2023-07-11:
looks like we need to rest the `LoggerdState` when get out of sync, including clearing the old packages that are in the queue.otherwise it will cause endless recursive calls.

deanlee, 2023-07-11:
already submitted two PRs to address the memory leak issues:

https://github.com/commaai/openpilot/pull/28873
https://github.com/commaai/openpilot/pull/28697

It may require a few more PRs to improve the stability of loggerd.

adeebshihadeh, 2023-07-12:
Do we need both? What's the smallest possible fix for the current issue?

deanlee, 2023-07-12:
#28873 is for the current issue.

the smallest possible quit and dirty fix could be something like this :https://github.com/commaai/openpilot/pull/28901

but I strongly recommend use #28873. the recursive call is very buggy.  we really should  refactoring loggerd instead of just keeping it working as it is.

adeebshihadeh, 2023-07-13:
I see, this is a big change to get in the upcoming release. I'll check out the fix you prefer after we ship this one.

PR #33453 (2025-03-03): loggerd: eliminate recursion from handle_encoder_msg
**Changes:**
1. **Refactored handle_encoder_msg:** Moved the encoding packet writing logic to a new function, `write_encode_data`, 
2. **Eliminated Recursive Calls:** Replaced recursive calls within `handle_encoder_msg` with direct calls to `write_encode_data` for writing queued packets in the new segment. This change prevents potential crashes by handling queued packets safely, even if they are out of sync with the encoderd.

~3. **Improved Memory Management:** Replaced raw pointers with std::unique_ptr for managing Message objects, improvi [... cut]
closed 2025-03-03 by adeebshihadeh