commaai/openpilot #28860: FPv2: skip duplicate queries from multiple brands

sshane, 2023-07-10:
We probably have a few queries for the same ECU address, response offset, and data that we can re-use for other brands when CAN fingerprinting or fuzzy fingerprinting, don't need to re-query. Should add them to the carFw list with each brand

adeebshihadeh, 2024-01-09:
Does this affect the best case? If not, don't think we care

sshane, 2024-01-09:
There could be some subtleties here, behavior could change in certain edge cases. Probably fine to not do anything special, like adding fake responses for duplicate requests (so we can see if ECU changes its response, or doesn't respond if the first one was too quick, etc.)

Plus with fuzzy fingerprinting now finishing when the first brand matches, the speed increase doesn't make as much sense.

closed 2024-01-09 by sshane