commaai/openpilot #28903: nav: hides settings when re-routing

sshane, 2023-07-12:
When you re-route in the nav settings view, it switches to the map, but I don't think it shows the map if the view is hidden

adeebshihadeh, 2023-07-16:
I don't understand this issue?

sshane, 2023-07-16:
This is much less of an issue since we hide the settings on interactive timeout now. It would hide the settings when you re-route, where I don't think it shows the map when you re-route and it's now showing.

sshane, 2023-07-24:
This is a simple fix, we can just move this line into the allow_open check which I think is more consistent. Any objections @incognitojam?

https://github.com/commaai/openpilot/blob/1776c16f9a61dbb1e322f15d0f1f3c0a9eb936d8/selfdrive/ui/qt/maps/map.cc#L195

Added in https://github.com/commaai/openpilot/pull/28884

closed 2023-07-25 by sshane