commaai/openpilot #28919: dmonitoringmodeld never started running

adeebshihadeh, 2023-07-13:
Likely stuck in the VisionIPC connect loop

```
6111bd7df5043394|2023-07-13--08-41-26
```

deanlee, 2023-07-14:
Checking.

adeebshihadeh, 2023-07-15:
Happened again with the logging and appears it's stuck in the DM model init 

```
batman:debug$ ./filter_log_message.py "00605c23b4af39f5|2023-07-14--21-16-49" 0 | grep selfdrive.modeld | grep -v manager.py | grep -v "\/modeld.cc"
[60.259997] selfdrive/modeld/runners/snpemodel.cc:36 - SNPEModel: loaded model with size: 3630942
[60.264494] selfdrive/modeld/runners/snpemodel.cc:36 - SNPEModel: loaded model with size: 4443461
[60.607753] selfdrive/modeld/runners/snpemodel.cc:89 - addInput: adding index 0: input_img
[60.608591] selfdrive/modeld/navmodeld.cc:47 - main: models loaded, navmodeld starting
[60.609128] selfdrive/modeld/navmodeld.cc:56 - main: connected with buffer size: 98304
```

I think it's either #28598 or some kind of race condition now that we're running the navmodel on the DSP as well.

deanlee, 2023-07-15:
https://github.com/commaai/openpilot/pull/28598 looks fine so far. it's more likely some kind of race condition.

adeebshihadeh, 2023-07-17:
Looks like it's a race condition that started after we enabled the navmodel.

navmodel enabled (#28448) on 6/16, first occurrence is 6/24. modeld refactor merged on 7/6.

deanlee, 2023-07-17:
The problem I've found so far is that `mapsd` exits without releasing `MapRenderer`, which can lead to deadlocks or some other undefined behavior with vipc/msgq after restarts.

deanlee, 2023-07-17:
mapsd will segfault on exit if create MapRenderer on stack:

>   MapRenderer m(get_mapbox_settings());
>   return app.exec();

```
corrupted size vs. prev_size in fastbins
Aborted (core dumped)
```

sshane, 2023-07-19:
Repro below. You also need to make boardd listen to `FakeIgnition`:

```python
import time
from cereal.messaging import SubMaster
from common.realtime import DT_MDL
from common.params import Params

def all_dead(managerState):
  # return if all processes that should be dead are dead
  return all([not p.running or p.shouldBeRunning for p in managerState.processes])

if __name__ == "__main__":
  params = Params()
  params.put_bool("FakeIgnition", False)

  sm = SubMaster(["driverStateV2", "managerState", "deviceState"])
  occurrences = 0
  loops = 0

  while 1:
    params.put_bool("FakeIgnition", True)
    sm.update(0)
    dmon_frame = None

    # print('Waiting for driverStateV2')
    st = time.monotonic()
    timeout = 15  # s

    # successful if we get 100 messages from dmonitoringmodeld (2s)
    while time.monotonic() - st < timeout:
      sm.update(0)
      time.sleep(DT_MDL)
      if sm.updated["driverStateV2"]:
        if dmon_frame is None:
          dmon_frame = sm.rcv_frame['driverStateV2']

        if (sm.rcv_frame['driverStateV2'] - dmon_frame) > (2 / DT_MDL):
          # print('Got driverStateV2! Exiting', sm.rcv_frame['driverStateV2'], dmon_frame)
          time.sleep(1)
          break
    else:
      occurrences += 1
      print('WARNING: timed out in 15s waiting for 40 messages from dmonitoringmodeld, occurrences:', occurrences, sm.rcv_frame['driverStateV2'], dmon_frame)
      print('CurrentRoute:', params.get('CurrentRoute'))

    # TODO: is there a better way? we can't check managerState immediately since it takes a while to get the ignition
    # wait for thermald to pick up ignition, then an update from managerState, and THEN it should be safe to check procs
    params.put_bool("FakeIgnition", False)
    while sm['deviceState'].started:
      sm.update(0)
      time.sleep(0.05)

    while not sm.updated['managerState']:
      sm.update(0)
      time.sleep(0.05)

    st = time.monotonic()
    while time.monotonic() - st < timeout:
      sm.update(0)
      time.sleep(0.1)
      if all_dead(sm['managerState']):
        # print('all dead')
        break
    else:
      print('WARNING: timed out waiting for processes to die!', time.monotonic() - st)
      time.sleep(5)

    loops += 1
    if loops % 120 == 0:
      print('Tries so far:', loops, 'occurrences:', occurrences)
```

sshane, 2023-07-20:
Added some logging in snpemodel.cc, here's what we got (note that there should be two of every log since navmodeld and dmonitoringmodeld both go through this path)

```python
batman@workstation-shane:~/openpilot/selfdrive/debug$ ./filter_log_message.py "8b787d8e2d5efdca|2023-07-20--03-56-48" 0 | grep \/modeld
[34191.163662] selfdrive/modeld/runners/snpemodel.cc:28 - SNPEModel: got runtime
[34191.225671] selfdrive/modeld/runners/snpemodel.cc:37 - SNPEModel: loaded model with size: 4443461
[34191.226534] selfdrive/modeld/runners/snpemodel.cc:41 - SNPEModel: snpe_builder
[34191.227161] selfdrive/modeld/runners/snpemodel.cc:43 - SNPEModel: !snpe
[34191.405491] selfdrive/modeld/runners/snpemodel.cc:37 - SNPEModel: loaded model with size: 3630942
[34191.406475] selfdrive/modeld/runners/snpemodel.cc:41 - SNPEModel: snpe_builder
[34191.407061] selfdrive/modeld/runners/snpemodel.cc:43 - SNPEModel: !snpe
[34191.672388] selfdrive/modeld/modeld.cc:209 - main: models loaded, modeld starting
[34191.878895] selfdrive/modeld/runners/snpemodel.cc:60 - SNPEModel: got snpe
[34191.879482] selfdrive/modeld/runners/snpemodel.cc:64 - SNPEModel: created output buffer
[34191.880911] selfdrive/modeld/runners/snpemodel.cc:67 - SNPEModel: debug print1
[34191.882317] selfdrive/modeld/runners/snpemodel.cc:72 - SNPEModel: debug print2
[34191.883093] selfdrive/modeld/runners/snpemodel.cc:74 - SNPEModel: debug print3
[34191.883740] selfdrive/modeld/runners/snpemodel.cc:81 - SNPEModel: debug print4
[34191.884317] selfdrive/modeld/runners/snpemodel.cc:83 - SNPEModel: debug print5
[34191.884877] selfdrive/modeld/runners/snpemodel.cc:85 - SNPEModel: debug print6
[34191.885416] selfdrive/modeld/runners/snpemodel.cc:97 - addInput: got input tensor names
[34191.885665] selfdrive/modeld/runners/snpemodel.cc:101 - addInput: got single tensor name
[34191.885877] selfdrive/modeld/runners/snpemodel.cc:103 - addInput: adding index 0: input_img
[34191.886082] selfdrive/modeld/runners/snpemodel.cc:106 - addInput: new debug print1
[34191.886389] selfdrive/modeld/runners/snpemodel.cc:111 - addInput: new debug print2
[34191.886609] selfdrive/modeld/runners/snpemodel.cc:119 - addInput: new debug print3
[34191.886820] selfdrive/modeld/runners/snpemodel.cc:126 - addInput: new debug print4
[34191.887035] selfdrive/modeld/runners/snpemodel.cc:129 - addInput: new debug print5
[34191.887251] selfdrive/modeld/runners/snpemodel.cc:132 - addInput: new debug print6
[34191.887462] selfdrive/modeld/navmodeld.cc:47 - main: models loaded, navmodeld starting
[34191.887690] selfdrive/modeld/navmodeld.cc:56 - main: connected with buffer size: 98304
[34192.574439] selfdrive/modeld/modeld.cc:226 - main: vision stream set up, main_wide_camera: 0, use_extra_client: 1
[34192.618304] selfdrive/modeld/modeld.cc:240 - main: connected main cam with buffer size: 4804608 (1928 x 1208)
[34192.619037] selfdrive/modeld/modeld.cc:244 - main: connected extra cam with buffer size: 4804608 (1928 x 1208)
[34192.691559] selfdrive/modeld/modeld.cc:173 - run_model: skipping model eval. Dropped -1 frames
```

adeebshihadeh, 2023-07-23:
Merged #29080, but I'd still like to fix this properly soon. Moving to the 0.9.5 milestone.

adeebshihadeh, 2023-08-24:
Closing since it doesn't seem worth spending more time on looking into this SNPE issue - the real fix is moving to tinygrad!

closed 2023-08-24 by adeebshihadeh