commaai/openpilot #28990: Ford: steering oscillation

incognitojam, 2023-07-17:
### Describe the bug

While driving straight on the highway at ~75mph the steering wheel began oscillating for 20 seconds before resolving itself.

### Which car does this affect?

Ford Focus 2019

### Provide a route where the issue occurs

e886087f430e7fe7|2023-07-16--21-08-19--11

### openpilot version

master

sshane, 2023-08-08:
Does slow ramp type help?

incognitojam, 2023-08-10:
I'll try it out next time I go on the highway. Did you try changing precise/comfortable?

blue-genie, 2024-01-12:
I'm hitting the same issue in my F-150, I know it's not formally supported, but trying to help to make ti better and be supported.

The current setting for `LatCtlRampType_D_Rq` for all Fords is set to `0`, `Slow Ramp speed`

Ford `LatCtlCurv_No_Actl` signal:
[image]

OP `LatCtlCurv_No_Actl` signal:
[image]

note that ford sends commands between 0.00006 and -0.00004, step size is 0.00002, over 40 seconds while OP adjust the signal between 0.0003 and -0.0003

I played with `steerActuatorDelay` starting form `0.0` to `0.2`(default value) on `0.05` increments, it had some improvement, but not to significant.

I'm more than happy to do the legwork, but maybe some pointers would be useful

adamgranted, 2024-02-01:
Same issue in my 2021 Explorer ST. Getting oscillations in both modes but experimental seems to be worse? Could be placebo. Let me know what I can do to help.