commaai/openpilot #29026: tools/replay: set NavDestination from the route

adeebshihadeh, 2023-07-19:


sshane, 2023-07-19:
I was thinking it would be better if we got rid of the param reading in map.cc and just have navRoute publish its navDestination. Similar thing to controlsd with `experimentalMode`, and plannerd with `longitudinalPersonality`.

deanlee, 2023-07-20:
I agree with @sshane . otherwise we also have to prevent the ui from fetching the NavDestination from the CommaApi during replay.

deanlee, 2023-07-20:
I agree with @sshane .this can simplifies the synchronizations, replay also works without modification.

adeebshihadeh, 2023-08-13:
Good point, this is definitely the right solution. Not sure why that's only broadcasted on change.

sshane, 2023-08-26:
@deanlee can you check how big navRoute is and make a PR to send it at a fixed frequency? I think there were some concerns about the size of the service inflating rlogs/qlogs

closed 2024-01-05 by adeebshihadeh