commaai/openpilot #29047: Subaru: New LKAS message

jnewb1, 2023-07-20:
### Describe the bug

Shane found that the new LKAS message on 22 forester actually seems to be a steering angle request.
[image]

Eyesight is still sending the old ES_LKAS message though, with the request bit set to zero. It might be possible to send that message instead, and block/disable the steering angle request.

### Which car does this affect?

Forester 22, and possibly all cars with the new C/D harness

### Provide a route where the issue occurs

NA

### openpilot version

NA

jnewb1, 2023-07-20:
b0cea2db2dd5474e|2023-07-19--20-53-52

tried old ES_LKAS message, EPS did not respond, even with the ALT message request bit set to zero. if you fully block ES_LKAS_ALT, you get an EPS fault.

jnewb1, 2023-07-20:
updated opendbc: https://github.com/commaai/opendbc/pull/904

jnewb1, 2023-07-24:
Have a 22 forester and a 23 outback running now with an angle controller. Needs tuning.

adeebshihadeh, 2023-07-24:
Moving to a discussion, they're better for things that aren't strictly issues