commaai/openpilot #29058: Radar Error

ReggieR3gg, 2023-07-20:
### Describe the bug

Randomly started occurring within the past few days can't figure out what's triggering this
[image]

### Provide a route where the issue occurs

d083dc54dae836f0|2023-07-20--05-55-48

### openpilot version

0.9.3

sshane, 2023-07-20:
Looks like something is wrong canState1:

[image]

The camera wants to stay engagd and continues to command acceleration, so doesn't seem like a car issue.

I would try a new USB-C cable in the mean time, but sounds like a possible issue and you should contact support: https://comma.ai/support

sshane, 2023-09-29:
Any updates @ReggieR3gg? Does this still happen?

ReggieR3gg, 2023-09-29:
Hi Shane, thankfully a new usb-c cord did the trick! Haven't had the error since

Sent via the Samsung Galaxy S22 Ultra 5G, an AT&T 5G smartphone

sshane, 2023-09-29:
Thanks for confirming!

closed 2023-09-29 by sshane