commaai/openpilot #29085: Ford F150 - Car Unrecognized - Iso-tp errors

sheaduncan, 2023-07-21:
### Describe the bug

Received Car Unrecognized - FW Query and Iso-tp errors 09136c309ba9461d|2023-07-21--16-18-33

### Which car does this affect?

Ford F150

### Provide a route where the issue occurs

09136c309ba9461d|2023-07-21--16-18-33

### openpilot version

.94

sshane, 2023-07-22:
can you get a drive on master?

sheaduncan, 2023-07-22:
Observation: I went for 2 drives before this error occurred. Here is one of those drives.
09136c309ba9461d|2023-07-21--17-11-46

When I went back out a few hours later it took the C3 to boot up longer than normal and from there on it was Car Unrecognized. Nothing had changed in-between that time.

Just reinstalled master and no issue, strange.
09136c309ba9461d|2023-07-21--20-02-05

What do you expect was the issue?

sshane, 2023-07-22:
Works fine on master, make sure you update your fork branch and try again. You can message me on Discord if it doesn't work again

sheaduncan, 2023-07-22:
Nope, I did see one of my restarts said "Communication error" though. I understand on your side that master works. However I'm still scratching my head about what it could of been. I won't spend too much time on it and just see if it happens again.

sheaduncan, 2023-07-22:
Fingerprint issues again but on master this time. It's when I start the vehicle and the C3 has to boot up. I powered cycled the vehicle and it registered the Fingerprint on the second one.

Route that experienced Car Unrecognized.
09136c309ba9461d|2023-07-22--09-30-18

sshane, 2023-07-24:
Can you put your device on WiFi to upload the route?

sheaduncan, 2023-07-28:
The rlog for that route was wiped when I moved branches, unfortunately. I've DMed you on Discord with some additional information.

sshane, 2023-07-31:
It appears that it is common for the ECUs to be sending diagnostic responses while starting up, so it's probable our requests are getting ignored.

sheaduncan, 2023-10-13:
> It appears that it is common for the ECUs to be sending diagnostic responses while starting up, so it's probable our requests are getting ignored sometimes. In the latest route you gave me though, it did respond to the engine query, but only in ISO-TP CAN FD mode, so we didn't parse it:
> 
> [image]
> 
> The panda PR adding CAN FD ISO-TP support should fix this: [commaai/panda#1524](https://github.com/commaai/panda/pull/1524)

How are you seeing this data? I've looked through the useradmin panel and haven't seen this data format.

sheaduncan, 2023-11-12:
I've implemented the UDS fix that @sshane  has a PR for, and while it does return more queries, my vehicle still fails to fingerprint most of the time. https://github.com/commaai/panda/pull/1524

Observation: No one with a C3X has this issue on the CAN-FD platform. Looking at all their queries, they return on bus 0 or bus 1. In comparison, most of my queries return on bus 4.

closed 2024-01-04 by sheaduncan