commaai/openpilot #29086: logmessaged: drops first few log messages for new processes

sshane, 2023-07-21:


sshane, 2023-07-21:
@deanlee want to check this out?

sshane, 2023-07-22:
Actually seems like it's process specific, navmodeld doesn't have any log messages dropped on startup but dmonitoringmodeld does.

deanlee, 2023-07-22:
it looks like `SNPEModel: loaded model with size: 4443461, models/dmonitoring_model_q.dlc` is alreay the first log in `dmonitoringmodeld`. Did you add any other logs before that?

sshane, 2023-07-22:
Yes, on my branch: https://github.com/commaai/openpilot/blob/93dcf2ee7dae9da1b1d9059a98a07c630c480f04/selfdrive/modeld/dmonitoringmodeld.cc#L49-L52

I also added more logs in the init function that didn't get saved.

deanlee, 2023-10-13:
This issue is most likely caused by loggerd starting after dmonitoringd:

https://github.com/commaai/openpilot/blob/818d29818fcb54f84e8fbb366fbe6913c70be2b9/selfdrive/manager/process_config.py#L55-L58

adeebshihadeh, 2024-01-05:
Is this a real issue or can we close?

closed 2026-09-12 by adeebshihadeh