commaai/openpilot #29104: OS Update Required - Fails

ns3284, 2023-07-23:
### Describe the bug

On master-ci, and device says OS update is required, but fails to update immediately after pressing install and tells me to reboot. Tried 2 different networks and consistently fails. 

### Provide a route where the issue occurs

NA

### openpilot version

0.9.4 latest

ns3284, 2023-07-23:
[image]
[image]
[image]

ridgewayer25, 2023-07-23:
Same issue, didn’t see this when I was opening report for same thing

Verylukyguy, 2023-07-23:
Same problem for me on master and nightly.

adeebshihadeh, 2023-07-23:
This was caused by adding an import to functools.cache in [gpio.py](https://github.com/commaai/openpilot/blob/12ed894362dd22513a21de6689a31511b791e37b/common/gpio.py#L1), which was new in Python 3.9. We extensively test the update, but that was added after the update was merged. I'm going to look into setting up a test for this. It's fixed by 5831f9e5a0d74a03a1b6ebc7b87f6d76d0fc38ae.

closed 2023-07-23 by adeebshihadeh

ns3284, 2023-07-23:
Is there an easy way to get the latest update? Haven't had the opportunity to play around with it, so I might be stuck in this update loop. SSH still work at this point? Just connect and do a git pull?

adeebshihadeh, 2023-07-23:
Yes, SSH still works and a git pull will fix (though it'll be more of a fetch & reset since master-ci is force pushed. You can also do a system reset by tapping on the screen as soon as it's booting up.