commaai/openpilot #29122: Masda Cx5 adaptive cruise control not working

satishmantri, 2023-07-24:
### Describe the bug

when i  engage the OpenPiolot on my Masda CX5 2022 adaptive cruise control not stopping for the cars in front of my car.

### Provide a route where the issue occurs

please check my device is add to my account

### openpilot version

0.9.3

satishmantri, 2023-07-24:
route https://connect.comma.ai/b36f9b30a01c8521/1690235466490/1690236742965
dongle id b36f9b30a01c8521

vanillagorillaa, 2023-07-25:
openpilot does not control the Adaptive Cruise Control functionality on your vehicle, this is left to the vehicles stock system to handle. This is not an issue with openpilot, but with your vehicle.

closed 2023-07-25 by vanillagorillaa

satishmantri, 2023-07-25:
Thank you for your response But this only happened after I added the comma3
into obd2 port and it was working fine before ?

I remember it was warning something in those lines while I was inverting
the comma into obd port

Can you please help me with this?

Thank you
Satish