commaai/openpilot #29131: HDC error Honda HRV eHEV (hybrid,radarless)

somkanae, 2023-07-25:
### Describe the bug

I use Honda hrv eHev 2022 (hybrid , radarless)  when openpilot engage HDC error show
[image]
[image]

### Provide a route where the issue occurs

1e4baee1aa2687a0|2023-07-22--15-46-12--0

### openpilot version

0.9.4

sshane, 2023-07-25:
User has tried new harness, but pandaStates and can hangs whenever the safety mode is changed:

1e4baee1aa2687a0|2023-07-22--03-05-42--0

[image]

adeebshihadeh, 2023-08-07:
The panda seems to be getting flooded with (presumably error) interrupts from bus 1. Can you get a route on master or master-ci? We have some new logging that would be helpful here.

sshane, 2023-08-07:
Have a recent route here, anything added since then? 1e4baee1aa2687a0|2023-07-23--15-39-57

eFiniLan, 2023-08-10:
1e4baee1aa2687a0|2023-08-10--21-21-37
this is on the master w/ "enable openpilot" off

sshane, 2023-08-24:
@briskspirit have you checked this out?

sshane, 2023-09-15:
@briskspirit any updates? user is still experiencing this

briskspirit, 2023-09-15:
I don't see any error flooding from CAN cores. Something else hangs panda (inside IRQ call) for that brief second. we need at least figure out which interrupt it hangs in. I will work on test branch for him today. @sshane might need your help on OP side.

sshane, 2023-09-16:
@somkanae can you give this branch a try and report back?

https://github.com/commaai/openpilot/tree/issue_29131

Installer URL: https://installer.comma.ai/commaai/issue_29131

somkanae, 2023-09-16:
After I installed op shown Car Unrecognized
[image][image]

sshane, 2023-09-18:
@briskspirit hmm can and pandaStates still drop out during fingerprinting.

sshane, 2023-09-29:
@briskspirit can you check out the new route on master?

briskspirit, 2023-09-29:
@sshane updated my panda branch to also report longest execution time for IRQ. Seems like in this case it isn't rate issue but execution time of some handler.
 You need to update your fork.

sshane, 2023-10-03:
@somkanae Hi, I updated the issue branch. Please pull, update submodules and try again!

somkanae, 2023-10-04:
[image]
[image]
[image]
[image]
@sshane After I installed issue branch Openpilot still unrecognized. 

https://connect.comma.ai/1e4baee1aa2687a0/1696381412303/1696381671763

1e4baee1aa2687a0|2023-10-04--08-03-32--0

sshane, 2023-10-10:
@somkanae please pull and try again

somkanae, 2023-11-20:
@sshane Please let me know how it goes.

sshane, 2024-04-15:
@somkanae is this still an issue?

somkanae, 2024-04-16:
@sshane Right now, I'm using the eFiniLan branch of Openpilot 0.9.6, https://github.com/eFiniLan/openpilot/tree/hrv_v2 which still has some errors.

somkanae, 2024-12-18:
I'm testing openpilot 0.9.8. After installation, there's a 'no panda' issue similar to what you suggested to try in issue #29131. Are there any updates or can you help analyze this further?