commaai/openpilot #29173: Restarting bridge. Error: rpc::rpc_error during call in function `get_sensor_token`

nathanoy, 2023-07-28:
### Describe the bug

Hello,

when executing the `bridge.py` it spams this error multiple times a second:
```
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
WARNING: Client API version     = 43b5e70-dirty 
WARNING: Simulator API version  = 0.9.13 
Restarting bridge. Error: rpc::rpc_error during call in function get_sensor_token
```

this occurs when executing either `./start_openpilot_docker.sh` or `./bridge.py` and `./launch_openpilot.sh`

My efforts to make the Client API version match the Simulator API version were unsuccessful.
 

### OS Version

Ubuntu 20.04.6 LTS

### openpilot version or commit

f911e5f02869475acfb634a058771390fb2273c7

### Additional info

# In Version 9.2.0 (release commit)
Hash: 89f68bf0cbf53a81b0553d3816fdbe522f941fa1
The same error occurs when executing `./start_openpilot_docker.sh` but not when executing `./bridge.py` and `./launch_openpilot.sh`.

This leads me to believe that the latest docker image of openpilot_sim might be a/the reason for this. (Please correct me if im wrong.)

haormj, 2023-07-28:
same question, ubuntu 22.04, run sim

dankanti20, 2023-07-30:
Hello guys, 
Please, I need your help on how to run this simulation. I'm novice.
I have already downloaded the dataset.

nathanoy, 2023-08-02:
After some more fooling around with the situation, it seems to me like the version change to Python 3.11 broke the client API compatibility.

In this pr #28843 the carla client version is defined to be exactly this:

https://github.com/commaai/openpilot/blob/21802265bec5ede0eaf4093ba0cd2c4c5be3b1af/pyproject.toml#L68C1-L68C1
```toml
carla = { url = "https://github.com/commaai/carla/releases/download/3.11.4/carla-0.9.14-cp311-cp311-linux_x86_64.whl", platform = "linux", markers = "platform_machine == 'x86_64'" }
```
Which seems to be made for carla version 0.9.14, yet the file `tools/sim/start_carla.sh` specifically uses `0.9.13`.
After updating `tools/sim/start_carla.sh` to use carla `0.9.14` the error disappears, only the version warning remains (makes sense).

This seems like a great opportunity to try make my first contribution.

closed 2023-08-02 by adeebshihadeh

Dragoadri, 2023-08-02:
THANKS SO MUCH THIS FIXED MY PROBLEM!