commaai/openpilot #29211: comma three stuck on loading screen and bootlooping

goodchoom, 2023-08-02:
### Describe the bug

Sat in my car this morning to find out my comma three was bootlooping. After I unplugged the power cable then plugged it back in, the device booted up normally and I was able to use openpilot with no issues. I had a similar occurrence on the same device a few days ago, except the screen was completely black while the fan was running, which was resolved the same way.

I’m on my second warranty replacement, and I’d rather not send it in for a third time if it turns out to be a hardware issue.

### Provide a route where the issue occurs

N/A

### openpilot version

ead064e981d928ffd51d6a8bd7153e90244b694d

goodchoom, 2023-08-02:
https://cdn.discordapp.com/attachments/1005992883452649483/1136317228657164379/IMG_6754.mov

Attached is a Discord link to the video of the comma three bootlooping as it won’t upload to GitHub for some reason.

adeebshihadeh, 2023-08-02:
Need a dongle ID to look into this.

goodchoom, 2023-08-02:
> Need a dongle ID to look into this.

5c49984c8efb9c9c

goodchoom, 2023-08-04:
It just hapoene

goodchoom, 2023-08-04:
It just happened again this morning. This time I noticed the light on the back of my device was red, and fading in and out.

adeebshihadeh, 2023-08-04:
The LED on the back isn't relevant here - that's just the panda. 

Unfortunately, this is pretty early in the startup, so there's not much logging. Can you provide any more details?
* Does it always reboot while in the spinner?
* How many loops have you let it go for? I suspect it'll just keep going indefinitely, but good to confirm

goodchoom, 2023-08-04:
> Does it always reboot while in the spinner?

Doesn't look like it.  From the video I recorded this morning, the large comma logo blinked at least three times before it got to the spinner, which is the first time I've seen this specific occurence happen.

https://github.com/commaai/openpilot/assets/85845724/57f425ba-2c6a-49f3-a278-154ad7709416

> How many loops have you let it go for? I suspect it'll just keep going indefinitely, but good to confirm

I assume this loops indefinitely as well. The longest I've let it go was for about a couple minutes.

adeebshihadeh, 2023-08-04:
Is it fine once it's booted? Fortunately, it's repeatable, so I can add some new logging to figure out what's going on.

goodchoom, 2023-08-04:
> Is it fine once it's booted? Fortunately, it's repeatable, so I can add some new logging to figure out what's going on.

Yes, once I unplug the device then plug it back in, I'm able to use openpilot with no issues. It seems like this happens every two days.

goodchoom, 2023-08-05:
https://github.com/commaai/openpilot/assets/85845724/ec5eb5e9-7916-4520-b527-e2f4e8ba0859

It just happened again this morning. The car is running and I can hear the fan, but the screen is completely blank.

[image]

Once again I unplugged the device and plugged it back in and was able to boot as usual, except this time I’m unable to use openpilot due to this error message. Rebooting does not resolve this issue. 

Regardless, I am now convinced my device is on its last legs and thus no longer comfortable with using it anymore.

goodchoom, 2023-08-05:
I understand the latest master-ci build resolves this error message. When I have the time I will factory reset and reinstall master-ci.

goodchoom, 2023-08-07:
I was able to factory reset and then re-install master-ci shortly after posting my previous comment. So far, I have yet to encounter any bootlooping or black screens, and have had no issues while using openpilot as well. However, I just realized I forgot to connect to my home Wi-Fi network since then. Could a problem with the Wi-Fi be a possible reason?

adeebshihadeh, 2023-08-08:
Unlikely, but possible. Since you can't repro anymore, I'm gonna leave this open until we add some logging. Let us know if you hit it again.

fleetingbytes, 2023-08-31:
> Need a dongle ID to look into this.

This happened to me too, yesterday. I was really scared, thought it's bricked. Dongle ID `44379c05b64fd7e3`. I turned it off and tried after a couple of hours, then it started normally.

It happened in this scenario:
- C3X is connected to a 5V USB power supply
- In the settings choose to make a reset to factory settings (wipe it)
- Low voltage warning appears (choice Power off or proceed at own risk)
- Choose Power off
- Not sure if I waited long enough for a clean power off
- Unplug, and within a few minutes connect it to the OBD-C in the car
- stuck in logo screen forever

goodchoom, 2023-09-15:
> Unlikely, but possible. Since you can't repro anymore, I'm gonna leave this open until we add some logging. Let us know if you hit it again.

Just encountered the black screen for the first time in weeks. This time I was on the optimus_prime branch 9d17fca4616bccf142657c57e9e4490d42190f97

goodchoom, 2023-09-26:
Encountered another instance of bootlooping after attempting to switch branches from nlp-driving to dummy_b4_model:

https://github.com/commaai/openpilot/assets/85845724/751fe2e1-55e3-4fee-98cf-e2c4237a27b8

Then, after unplugging the device and plugging it back in, openpilot failed to build:

[image]

I was able to factory reset and reinstall openpilot on the master-ci branch, but this took much longer than expected. The progress bar was slow to move and got stuck several times in the process. It took about half an hour to download and install with the SIM card’s LTE connection.

goodchoom, 2023-10-31:
I just got back from a month long trip overseas, the comma three unplugged and stored neatly inside its box during that time. When I plugged in my comma three I was once again greeted with the openpilot failed to build error. I performed the usual factory reset then attempted to install the nlp-driving branch.

Everything seemed to go fine until the installer froze  at 91%. After waiting several minutes to no avail I unplugged the device, plugged it back in, and tried again. This time the loading bar went crazy for a few seconds, but openpilot successfully installed after a couple of minutes.

openpilot is up and running right now but at this point I am concerned I am dealing with a possible hardware issue.

mkos11, 2023-11-25:
It seems like you're experiencing issues with your Comma Three device bootlooping. Here are some general steps you can take to troubleshoot and potentially resolve the issue:

1. **Check for Software Updates:**
   Ensure that your openpilot software is up to date. Sometimes, bugs are fixed in newer software releases.

2. **Reinstall openpilot:**
   Reinstalling the openpilot software might help if there are any corrupted files causing the bootloop. Follow the recommended procedure for reinstalling openpilot on your Comma Three device.

3. **Review Logs:**
   Check the logs generated by openpilot for any error messages or warnings. This information can provide insights into what might be causing the bootloop.

4. **Community Forums:**
   Visit the Comma Three community forums or support channels. Other users might have faced similar issues and could provide valuable insights or solutions.

5. **Contact Support:**
   If the issue persists, reach out to Comma AI's official support channels. They might have specific troubleshooting steps or be aware of any common issues related to bootlooping on the Comma Three.

6. **Hardware Inspection:**
   Since you've already had a warranty replacement, it's essential to consider the possibility of a hardware issue. Check for any physical damage or loose connections that might be contributing to the problem.

7. **Factory Reset:**
   If possible, perform a factory reset on your Comma Three device. This will revert the device to its default settings, eliminating any potential software glitches.

8. **Documentation:**
   Refer to the official documentation for your Comma Three device. There might be specific instructions or troubleshooting steps provided by Comma AI that can help resolve bootloop issues.

Remember that these are general suggestions, and the specific solution might depend on the details of your situation. If the problem persists, reaching out to Comma AI support is crucial, especially considering that your device is still under warranty. They can provide personalized assistance and guide you through the troubleshooting process.

closed 2024-01-05 by adeebshihadeh