commaai/openpilot #29213: Lexus NX-300h hybird 2020 "Radar Error: Restart the Car"

Abanggo, 2023-08-02:
### Describe the bug

1. The symptom occurred and I removed the installed one.
2. After installing Comma 3 on the vehicle and installing master-ci (1cb188 / Aug 03) in an initialized state, all preliminary preparations have been completed.
3. Start riding and the adjustment is complete.
4. When C3 is operated in a vehicle that has been prepared for everything, the error message "Radar Error: Restart the Car" is displayed and the operation does not work.
    - Even if this operation is performed repeatedly, the symptom is the same, "Radar Error: Restart the Car" error message is displayed and OPENPILOT does not operate normally.
5. Even after rebooting, this symptom still occurs.

[image]
[image]

### Which car does this affect?

Lexus NX-300h hybird 2020

### Provide a route where the issue occurs

9a1f34e34fb917f2|2023-08-03--06-03-57

### openpilot version

openpilot 0.9.5 / master-ci /1cbe188 / Aug 03

### Additional info

1. There is also a post on Discord that this error message appears in some models.
2. If you remove COMMA3 from Lexus NX-300h hybird 2020 and drive, RADAR works very normally.
3. "Radar Error: Restart the Car" error occurs equally in both Dargonpilot / Sunnypilot, which are sub-forks.

adeebshihadeh, 2023-08-02:
@sshane can you check this out?

sshane, 2023-08-02:
canCore1 is continually being reset. Related to https://github.com/commaai/panda/pull/1502?

[image]

adeebshihadeh, 2023-08-02:
No, that should only apply to H7, which is only used on the 3X and red panda.

sshane, 2023-08-18:
hey @Abanggo does this still happen for you?

Abanggo, 2023-08-18:
> 여기요@Abanggo아직도 이런 일이 발생합니까?

Haven't been able to use it until now.

It has never worked properly since I was notified.
When I start the car, the same error message comes out and it is unusable.

I can't use it normally, I'm just staring at C3, and I'm checking github every day to see if the problem has been resolved.

It seems that it is possible to solve the problem completely by exchanging it only when it is determined whether it is a C3 problem or a SW problem.

COMMA AI should confirm that it is a HW problem so that it can be exchanged, and I think that if it is a SW problem, we can wait.

please help me what to do

sshane, 2023-08-18:
Recent routes still show the canCore1 with lots of total errors and can core resets. I would open a support ticket at https://comma.ai/support

cc: @adeebshihadeh in case you want the user to try anything with panda

Abanggo, 2023-08-19:
Thanks for the notification, I deleted the previously installed program, updated to the latest version of openpilot master-ci, and as a result of the test drive, the problem was resolved and it works well.
thank you

sshane, 2023-08-19:
It looks like the errors are still happening, but radard is ignoring the missing CAN traffic. Requested rlogs, can you put your device on wifi?

Abanggo, 2023-08-19:
> 오류가 계속 발생하는 것처럼 보이지만, radard는 누락된 CAN 트래픽을 무시하고 있습니다. 요청된 rlogs, 장치를 Wi-Fi에 둘 수 있습니까? 여전히 지원팀에 문의하세요. **이 문제는 여전히 발생** 하지만 버그로 인해 문제가 가려지고 있습니다. **레이더를 사용하고 있지 않습니다.**

9a1f34e34fb917f2|2023-08-18--14-08-34--0

9a1f34e34fb917f2|2023-08-18--05-20-24--0

9a1f34e34fb917f2|2023-08-18--05-24-48--0

9a1f34e34fb917f2|2023-08-18--14-33-12--0

9a1f34e34fb917f2|2023-08-18--14-46-36--0

I uploaded the log file I drove for the MASTER-CI test.

Abanggo, 2023-08-19:
As long as there is the latest update, we will update MASTER-CI additionally and check again whether it works normally.
thank you

Abanggo, 2023-08-20:
> 최근 경로는 여전히 총 ​​오류가 많은 canCore1을 표시하고 코어 재설정을 할 수 있습니다. https://comma.ai/support 에서 지원 티켓을 엽니다.[](https://comma.ai/support)
> 
> 개통 전 포크가 아닌 **스톡 openpilot** 에서 최근 경로를 확인하십시오 .
> 
> 참조:@adeebshihadeh사용자가 팬더로 무엇이든 시도하기를 원하는 경우

I was trying to upload an error message, but I accidentally received the wrong one in progress.
I'm sorry.
bring it back to life

Abanggo, 2023-08-20:
> It looks like the errors are still happening, but radard is ignoring the missing CAN traffic. Requested rlogs, can you put your device on wifi? Make sure to contact support still, **this issue is still happening**, but a bug is masking it. **You are not using your radar.**

0.9.5/master-ci/ed9f630/Aug 18

9a1f34e34fb917f2|2023-08-20--08-59-16--1

The log file has been uploaded.

1. TAKE CONTROL IMMEDIATELY Radar Error: Restart the Car error message screen display
     Error message displayed on the red background screen: Capturing failed while driving (displayed for a while and then disappeared)
    - Sudden occurrence during normal use of C3 openpilot
    - If ACC is re-operated afterwards, it will operate normally and continue to be used.
      * Estimated error occurrence time: between 2023.08.20.09:08 and 2023.08.20.09:10

2. The vehicle was driven in the afternoon, but it operated normally without any errors.

Abanggo, 2023-08-20:
[image]

This is the same mere message as the capture posted by someone else.
please refer to this

Abanggo, 2023-08-20:
openpilot 0.9.5 / master-ci / 5d0b7ef / Aug 21

The openpilot Unavailable error occurred again.
The log file has been uploaded.

9a1f34e34fb917f2|2023-08-21--05-30-18--0

9a1f34e34fb917f2|2023-08-21--05-39-02--0

[image]

sshane, 2023-08-21:
Contact support at https://comma.ai/support

Abanggo, 2023-08-21:
> https://comma.ai/support

Thank you for your help.
We will contact you at https://comma.ai/support.

sshane, 2023-08-21:
Have you tried a new USB cable?

Abanggo, 2023-08-22:
If the one I have doesn't meet the conditions, I can immediately buy one
and test it.

I will replace the USB-C GEN II cable and test it for about 2 days and let
you know the result.

Thank you for your interest.

Translated with google translator.

Abanggo, 2023-08-22:
I replaced the C3 cable with a new generic USB-C GEN II cable and tested it.
As a result, all problems in the meantime have been resolved and it works normally.
I've been test-driving dozens of times and no problems have occurred.

Is there any problem in using C3 even if I use USB-C GEN II cable continuously?

Thank you for your interest and response.
thank you

sshane, 2023-08-23:
Great! Thanks for your patience.

closed 2023-08-23 by sshane