commaai/openpilot #29224: Sudden Deceleration From Stop

dingyifei, 2023-08-03:
### Describe the bug

Sudden braking when the gas pedal is pressed from a stop and OP engaged. 

The issue occurred in Segment 26.

Previous conversation for reference

```
you were engaged down to a standstill, and successfully went into standstill (a specific state of the ACC subsystem)
[4:23 AM]
you then pressed the gas pedal, and you appear to have disengage on gas enabled
[4:24 AM]
the drivetrain/stopping coordinator will drop the EPB if ACC is disengaged while at standstill
[4:24 AM]
we actually don't have control over that
it's a feature with the car, that I'll have to think about how to deal with
[4:27 AM]
if you were to use stock ACC to follow someone to a stop, and while stopped, press the cancel button, the EPB would drop to prevent rollaway. this is by design.
[4:27 AM]
there are some different ACC control states, which I don't use right now, that are meant for use when the driver is using accel to override. it may be time to do that.
[4:28 AM]
right now this looks to the car like a disengage, and I probably need to show it as driver override to avoid the EPB drop
[4:28 AM]
there are some other nuances to why I'm not doing so now, and I will have to think about them
[4:30 AM]
TSK-side handling of the EPB is a safety thing, like, if the ACC radar suddenly failed/broke/crashed while you were behind a stopped car, the process of dropping the EPB is automatic and taken care of by the TSK independent of the ACC radar
[4:31 AM]
that's why EPB is required for stop-and-go cars, because the driver may not be fully attentive if the car has been at standstill for a while, so the car needs to safely handle the case of radar failure
```

### Which car does this affect?

VW Tiguan R-Line 4Motion 2020 (US)

### Provide a route where the issue occurs

4e144f64a2535890|2023-03-27--18-14-02--26

### openpilot version

master-ci V0.9.1

### Additional info

@jyoung8607

sshane, 2023-11-15:
Putting it in the override state and sending 0/inactive accel sounds like the plan.

jyoung8607, 2023-12-21:
Currently having some end users test a potential fix.

LANdrew12, 2024-07-01:
@jyoung8607 Has the test been working? I'm not sure if its fixed since I never accelerate from stops because now I intervene at every stop to prevent it.