commaai/openpilot #29231: athenad: reconnect can take longer expected

sshane, 2023-08-04:
The keep alives work as expected if the `ws_send` thread isn't started or it doesn't send any data, but in the test_athena_ping.py test, it sends three frames of usual startup data, which causes the timeout for `websocket._exceptions.WebSocketConnectionClosedException` to be roughly 2x what is set for `TCP_USER_TIMEOUT`.

Below are various `TCP_USER_TIMEOUT` with the reconnection test, and the amount of time it takes to go from [here](https://github.com/commaai/openpilot/blob/8332abf015ad2d6a962ed08c991dbd8099ecec58/selfdrive/athena/tests/test_athenad_ping.py#L85) to [here](https://github.com/commaai/openpilot/blob/8332abf015ad2d6a962ed08c991dbd8099ecec58/selfdrive/athena/athenad.py#L739).

```
TCP_USER_TIMEOUT=1000, 15743.990427962-15746.255553691=-2.2651257289999194
TCP_USER_TIMEOUT=5000, 15622.371322102-15630.351854196=-7.9805320939995
TCP_USER_TIMEOUT=7500, 15520.788894868-15536.673009143=-15.884114275000684
TCP_USER_TIMEOUT=10000, 15787.612519864-15803.452668979=-15.840149114999804
TCP_USER_TIMEOUT=16000, 15827.883680637-15859.135080651=-31.251400013999955
```

sshane, 2023-08-04:
Removing `TCP_USER_TIMEOUT` (setting to 0) sets it to a high kernel default it looks like, so it then relies on the ping timeout of 90 seconds to reconnect, so we'll want to keep that. Perhaps halve the value? I'm not sure why it's leading to 2x connection times when the keep alives are all set to 1 second.

jyoung8607, 2023-08-04:
If you want reliable connectivity without leaving-home disruptions, with careful socket binding, you can keep Athena on LTE (if available) at all times, and ship the bulk stuff over the least cost available transport.

adeebshihadeh, 2024-11-07:
Haven't noticed this in a long time

closed 2024-11-07 by adeebshihadeh