commaai/openpilot #29246: threex: didn't boot

sshane, 2023-08-05:
Device was connected with serial. I had just switched the branch to master and typed `sudo reboot`, then in serial all I see is `logout`. I disconnect serial (Ctrl+C), try to reconnect, but it showed `panda disconnected!` and no output. After waiting a minute with nothing displayed on the screen or serial, I unplugged, waited a few seconds, and plugged it back in.

Fan started up, nothing in serial, nothing on display. Unplugged, re-plugged to jungle now, fan stayed at max for a bit longer, but nothing still from display.

panda light was slow blinking red when I checked.

Finally, after plugging back into serial I got output and it showed the bootsplash and booted up successfully.

Bootlog right after this happened: bootlogs/297b4b460f361603/2023-08-05--01-05-27.bz2

sshane, 2023-08-05:
After it started, it looks like it disconnected from serial again and won't show any more output:

```python
[    9.826809] rmt_storage:INFO:main: Done with init now waiting for messages!
[    9.827499] ipa ipa3_uc_state_check:289 uC is not loaded
[   10.433533] samsung_i2c_touchpanel 2-0017: ss_color_cal_read: checksum 0x0 did not match expected 0xb8
[   10.447070] configfs-gadget a600000.dwc3: failed to start g1: -19
[   10.563559] samsung_i2c_touchpanel 2-0017: ss_color_cal_read: checksum 0x0 did not match expected 0xb8
[   10.843692] samsung_i2c_touchpanel 2-0017: ss_color_cal_read: checksum 0x0 did not match expected 0xb8

Ubuntu 20.04.5 LTS tici ttyMSM0

tici login: panda disconnected!
no pandas found
no pandas found
/home/batman/openpilot/panda/python/__init__.py:430: RuntimeWarning: found device with panda descriptors but invalid serial: ?
  warnings.warn(f"found device with panda descriptors but invalid serial: {serial}", RuntimeWarning)
no pandas found
no pandas found
no pandas found
no pandas found
no pandas found
no pandas found
no pandas found
no pandas found
no pandas found
panda disconnected!
USB connect error
Traceback (most recent call last):
  File "/home/batman/openpilot/panda/python/__init__.py", line 392, in usb_connect
    handle.claimInterface(0)
  File "/home/batman/.pyenv/versions/3.11.4/lib/python3.11/site-packages/usb1/__init__.py", line 1146, in claimInterface
    mayRaiseUSBError(
  File "/home/batman/.pyenv/versions/3.11.4/lib/python3.11/site-packages/usb1/__init__.py", line 127, in mayRaiseUSBError
    __raiseUSBError(value)
  File "/home/batman/.pyenv/versions/3.11.4/lib/python3.11/site-packages/usb1/__init__.py", line 119, in raiseUSBError
    raise __STATUS_TO_EXCEPTION_DICT.get(value, __USBError)(value)
usb1.USBErrorBusy: LIBUSB_ERROR_BUSY [-6]
^Cbatman@workstation-shane:~/openpilot/panda/tests$ ./som_debug.sh
```

adeebshihadeh, 2023-08-05:
What does "after plugging back into serial" mean? How did you get serial and power separately? 

> I unplugged, waited a few seconds, and plugged it back in.

Disconnecting power for a few seconds isn't gonna do anything with the supercaps fully charged, except kill and restart the fan since it's not on the supercaps.

> After it started, it looks like it disconnected from serial again

It's normal for it to disconnect after booting since openpilot reflashes and resets the panda on startup.

sshane, 2023-08-05:
I mean I unplugged the power cable which was going into serial. When I plugged it into the jungle, it was disconnected for about 15 seconds

adeebshihadeh, 2023-08-16:
Closing as this isn't really actionable

closed 2023-08-16 by adeebshihadeh