commaai/openpilot #29285: macOS: plotjuggler crashes on stream

royjr, 2023-08-08:
### Describe the bug

When using the `./tools/plotjuggler/juggle.py --stream` command, PlotJuggler successfully launches. However, after clicking the "Start" stream button and confirming the subsequent popup, PlotJuggler unexpectedly crashes.

[image]

```
(openpilot-py3.11) blah@Someoness-MacBook-Pro openpilot % ./tools/plotjuggler/juggle.py --stream
qt.qpa.fonts: Populating font family aliases took 47 ms. Replace uses of missing font family "Ubuntu" with one that exists to avoid this cost. 
Loading compatible plugins from directory:  "/Users/blah/github/openpilot/tools/plotjuggler/bin"
"libDataLoadRlog.dylib is a DataLoader plugin"
Loading schema: "/Users/blah/github/openpilot/cereal/log.capnp"
"libDataStreamCereal.dylib is a DataStreamer plugin"
Number of plugins loaded:  2 

Loading compatible plugins from directory:  "/Users/blah/Library/Application Support/PlotJuggler"
Number of plugins loaded:  0 

Using ZMQ backend!
Assertion failed: (port >= 0), function get_port, file impl_zmq.cc, line 22.
Stack trace (most recent call last):
#14   Object "QtCore", at 0x1025d0aef, in QMetaObject::activate(QObject*, QMetaObject const*, int, void**) + 3135
#13   Object "plotjuggler", at 0x1009bf09b, in MainWindow::qt_metacall(QMetaObject::Call, int, void**) + 99
#12   Object "plotjuggler", at 0x100a18e17, in MainWindow::on_buttonStreamingStart_clicked() + 187
#11   Object "plotjuggler", at 0x100a0d04f, in MainWindow::startStreamingPlugin(QString) + 587
#10   Object "libDataStreamCereal.dylib", at 0x11a2f1053, in DataStreamCereal::start(QStringList*) + 1007
#9    Object "libDataStreamCereal.dylib", at 0x11a3123d7, in SubSocket::create(Context*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, bool, bool) + 139
#8    Object "libDataStreamCereal.dylib", at 0x11a312e3f, in ZMQSubSocket::connect(Context*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, bool, bool) + 307
#7    Object "libDataStreamCereal.dylib", at 0x11a313113, in get_port(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>) + 431
#6    Object "libDataStreamCereal.dylib", at 0x11a3dc783, in get_port(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>) (.cold.1) + 39
#5    Object "libsystem_c.dylib", at 0x18e774e43, in __assert_rtn + 271
#4    Object "libsystem_c.dylib", at 0x18e775ae7, in abort + 179
#3    Object "libsystem_pthread.dylib", at 0x18e867c27, in pthread_kill + 287
#2    Object "libsystem_platform.dylib", at 0x18e896a23, in _sigtramp + 55
#1    Object "plotjuggler", at 0x100aa4c27, in backward::SignalHandling::sig_handler(int, __siginfo*, void*) + 19
#0    Object "plotjuggler", at 0x100aa4c77, in backward::SignalHandling::handleSignal(int, __siginfo*, void*) + 51
```

### OS Version

Ventura 13.4.1

### openpilot version or commit

master

### Additional info

It does not matter if the target ip is valid or if the bridge is started on the device.

adeebshihadeh, 2023-08-24:
@fredyshox can you look into this?

fredyshox, 2023-08-25:
I've updated PlotJuggler binaries for all platforms, streaming should work now. You can update using `tools/plotjuggler/juggle.py --install`

fredyshox, 2023-08-25:
Fixed with https://github.com/commaai/PlotJuggler/pull/58

closed 2023-08-25 by fredyshox