commaai/openpilot #29319: investigate + test build time

adeebshihadeh, 2023-08-11:
Keeping build time low is very important, and we have nothing that enforces this right now.
- [ ] tool to profile build time
- [ ] regression test
- [ ] make it faster?

adeebshihadeh, 2023-08-11:
Does mold speed things up? https://github.com/rui314/mold

closed 2024-02-02 by adeebshihadeh

MarinkoMagla, 2024-02-12:
Hi,

A few days before the issue was closed, I experimented with using the lld linker (https://github.com/llvm-mirror/lld) and optimization flags, and did see notable improvements in build time.

However, upon checking the closed issue, I couldn't spot any optimizations being implemented or any mention of build time improvements.

It might be worth revisiting this issue and discussing potential optimizations further.

Cheers!