commaai/openpilot #29335: bridge.py error in docker

PenNameLuXun, 2023-08-11:
### Describe the bug

root@docker-desktop:/openpilot/tools/sim# ./bridge.py
Traceback (most recent call last):
  File "/openpilot/tools/sim/./bridge.py", line 25, in <module>
    from tools.sim.lib.can import can_function
  File "/openpilot/tools/sim/lib/can.py", line 27, in <module>
    cp = get_car_can_parser()
         ^^^^^^^^^^^^^^^^^^^^
  File "/openpilot/tools/sim/lib/can.py", line 26, in get_car_can_parser
    return CANParser(dbc_f, signals, checks, 0)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "opendbc/can/parser_pyx.pyx", line 30, in opendbc.can.parser_pyx.CANParser.__init__
TypeError: __init__() takes at most 3 positional arguments (4 given)

my command is just follow https://github.com/commaai/openpilot/blob/master/tools/sim/README.md
when i run `./start_openpilot_docker.sh`,the error happened.
anyone can help?plz.

### OS Version

ubuntu22

### openpilot version or commit

4cad0a035dd8c78e0c44dc346708f2166cccb6c7

bvss89, 2023-08-11:
Same error for me as well. But running without docker.

adeebshihadeh, 2023-08-11:
@sshane can you fix? why didn't at least the linter catch this?

sshane, 2023-08-11:
pylint also doesn't check the wrong argument signature, but it does at least catch the unused `checks`, ruff doesn't.

```python
tools/sim/lib/can.py:21:2: W0612: Unused variable 'checks' (unused-variable)
```

sshane, 2023-08-11:
I couldn't find any way to do automatic function signature checking on imported Cython code, `stubgen` doesn't support it, and neither does pylint, ruff, or mypy that I could find. We can add manual stubs for the CANParser, but mypy doesn't check functions without any typing already in them, so it would not have caught this.

closed 2023-08-11 by sshane

PenNameLuXun, 2023-08-12:
It worked now, thanks.