commaai/openpilot #29344: static analysis for Cython modules

sshane, 2023-08-11:


belimedo, 2024-02-12:
Hi @sshane, I'm interested in working on this issue. 

As I'm new to the whole openpilot community, could you help me out and give a bit broader explanation regarding the requirements for this static analysis? 

For example. what type of static analysis are you looking for (syntax, semantics, data flow, etc.)? Is this something that is supposed to run as a pre-commit hook test, as a standalone test, or as a CI test (where in the project tree should I place this test)?

Thanks in advance!

jnewb1, 2024-02-19:
the linting is useful but this issue was for adding type hints to the cython modules, so you can get autocomplete for these modules

belimedo, 2024-02-19:
> the linting is useful but this issue was for adding type hints to the cython modules, so you can get autocomplete for these modules

Ahhhh, alright. Got it now. So something like this in this discussion on the Cython GH: https://github.com/cython/cython/issues/2587. I will take a deeper look at it, but so far it seems that this tool might solve what you need: https://github.com/Vizonex/CyStub 

If you can, please elaborate a bit more on what you are looking for, and if I understood it correctly now. Thanks in advance!

jnewb1, 2024-02-19:
I think you understand correctly, we'd like to be able to make stubs for these modules to get autocomplete

teodoramilanovic, 2024-04-22:
Hi @sshane, @jnewb1, @adeebshihadeh, I think I managed to improve one of the existing parsers to fit your needs, please take a look at this PR - https://github.com/commaai/openpilot/pull/32274

PR #31491 (2024-02-19): add Cython static analysis
**Description**

Fixes #29344

In this PR, I added [cython-lint](https://github.com/MarcoGorelli/cython-lint) for static analysis of Cython code. I also run the hook and fixed all of the errors this checker found.

@sshane, @adeebshihadeh If you can, please take a look at this PR, let me know if you wanted something else to be done for this static analysis. I ignored the same directories as `cpplint` checker ignores, and the only contraint I added was the line length (so that it also matches `cpplint` one).

Also, in the current master, file `selfdrive/ui/qt/maps/map [... cut]