commaai/openpilot #29370: macOS: update_translations.py not working

royjr, 2023-08-13:
### Describe the bug

```
(openpilot-py3.11) blah@Someoness-MacBook-Pro openpilot % ./selfdrive/ui/update_translations.py
sh: lupdate: command not found
Traceback (most recent call last):
  File "/Users/blah/github/openpilot/./selfdrive/ui/update_translations.py", line 53, in <module>
    update_translations(args.vanish, args.plural_only)
  File "/Users/blah/github/openpilot/./selfdrive/ui/update_translations.py", line 42, in update_translations
    assert ret == 0
           ^^^^^^^^
AssertionError
```

trying to use `third_party/qt5/larch64/bin/lupdate` results in..

```
(openpilot-py3.11) blah@Someoness-MacBook-Pro openpilot % ./selfdrive/ui/update_translations.py                                                                              
sh: /Users/blah/github/openpilot/third_party/qt5/larch64/bin/lupdate: cannot execute binary file
Traceback (most recent call last):
  File "/Users/blah/github/openpilot/./selfdrive/ui/update_translations.py", line 54, in <module>
    update_translations(args.vanish, args.plural_only)
  File "/Users/blah/github/openpilot/./selfdrive/ui/update_translations.py", line 43, in update_translations
    assert ret == 0
           ^^^^^^^^
AssertionError
```

### OS Version

Ventura 13.4.1

### openpilot version or commit

master

sshane, 2023-08-14:
Is there a dev version of qt5 you can install? Does Linguist exist on your machine?

eFiniLan, 2023-08-16:
one way you can try is install it via qt maintenance tool

eFiniLan, 2023-08-16:
@sshane is it ok to attach the bin files here?

sshane, 2023-08-16:
We have binaries for Darwin in third_party, so should be safe to add them if they work

sshane, 2023-08-18:
@fredyshox can you see about getting those binaries in openpilot?

fredyshox, 2023-08-18:
qt@5 tools should be already installed on your system if you used `mac_setup.sh`.  you can check if they're present at`$(brew --cellar)/qt@5/*/bin`

It's possible that brew didn't link them. Try running `brew link qt@5` to regenerate links.

royjr, 2023-08-18:
```
----   OPENPILOT SETUP DONE   ----
Open a new shell or configure your active shell env by running:
source /Users/blah/.zshrc

blah@Someoness-MacBook-Pro openpilot % source /Users/blah/.zshrc

blah@Someoness-MacBook-Pro openpilot % $(brew --cellar)/qt@5/*/bin    
zsh: permission denied: /opt/homebrew/Cellar/qt@5/5.15.10/bin

blah@Someoness-MacBook-Pro openpilot % sudo !!
sudo $(brew --cellar)/qt@5/*/bin
Password:
sudo: /opt/homebrew/Cellar/qt@5/5.15.10/bin: command not found

blah@Someoness-MacBook-Pro openpilot % brew list
==> Formulae
abseil		git-lfs		librist		m4		readline
aom		glfw		libsamplerate	mbedtls		rubberband
aribb24		glib		libsndfile	mpdecimal	sdl2
autoconf	gmp		libsodium	mpfr		six
brotli		gnutls		libsoxr		mpg123		snappy
bzip2		graphite2	libtasn1	nettle		speex
ca-certificates	harfbuzz	libtiff		opencore-amr	sqlite
cairo		highway		libtool		openexr		srt
capnp		icu4c		libunibreak	openjpeg	svt-av1
catch2		imath		libunistring	openssl@3	swig
cjson		isl		libusb		openssl@3.0	tesseract
cmake		jpeg-turbo	libvidstab	opus		theora
coreutils	jpeg-xl		libvmaf		p11-kit		tinyxml2
cppcheck	lame		libvorbis	pango		unbound
dav1d		leptonica	libvpx		pcre		webp
eigen		libarchive	libx11		pcre2		x264
ffmpeg		libass		libxau		pixman		x265
flac		libb2		libxcb		pkg-config	xorgproto
fontconfig	libbluray	libxdmcp	portaudio	xvid
freetype	libevent	libxext		protobuf	xz
frei0r		libidn2		libxrender	protobuf-c	z3
fribidi		libmpc		little-cms2	pyenv		zeromq
gcc@12		libnghttp2	llvm		python@3.11	zimg
gettext		libogg		lz4		qt@5		zlib
giflib		libpng		lzo		rav1e		zstd

==> Casks
appcleaner		gcc-arm-embedded

blah@Someoness-MacBook-Pro openpilot % brew link qt@5
Linking /opt/homebrew/Cellar/qt@5/5.15.10... 506 symlinks created.

If you need to have this software first in your PATH instead consider running:
  echo 'export PATH="/opt/homebrew/opt/qt@5/bin:$PATH"' >> ~/.zshrc

blah@Someoness-MacBook-Pro openpilot % sudo $(brew --cellar)/qt@5/*/bin         
sudo: /opt/homebrew/Cellar/qt@5/5.15.10/bin: command not found

blah@Someoness-MacBook-Pro openpilot % echo 'export PATH="/opt/homebrew/opt/qt@5/bin:$PATH"' >> ~/.zshrc

blah@Someoness-MacBook-Pro openpilot % sudo $(brew --cellar)/qt@5/*/bin                                 
sudo: /opt/homebrew/Cellar/qt@5/5.15.10/bin: command not found
```

royjr, 2023-08-19:
Running `/opt/homebrew/opt/qt@5/bin/lupdate` does work

```
blah@Someoness-MacBook-Pro ~ % /opt/homebrew/opt/qt@5/bin/lupdate
Usage:
    lupdate [options] [project-file]...
    lupdate [options] [source-file|path|@lst-file]... -ts ts-files|@lst-file

lupdate is part of Qt's Linguist tool chain. It extracts translatable
messages from Qt UI files, C++, Java and JavaScript/QtScript source code.
...
```

royjr, 2023-08-19:
Running `brew link qt@5` or `echo 'export PATH="/opt/homebrew/opt/qt@5/bin:$PATH"' >> ~/.zshrc` does the trick! I assume it just needs to be added to `openpilot_env.sh`

fredyshox, 2023-08-19:
Thats good. Gonna add it here: https://github.com/commaai/openpilot/pull/29456

closed 2023-08-24 by adeebshihadeh