commaai/openpilot #29380: CI <10 mins again

adeebshihadeh, 2023-08-13:
- [ ] docs build ~15 mins
This one builds openpilot every time without a SCons cache. Fixing that should bring it down to a few minutes.

- [ ] test car models ~15 mins
This has grown with the number of supported cars. `pytest --durations=0` should show where the time is being spent. 

Running on BuildJet will help here, but want to make sure it's well optimized before we do that.

- [ ] build mac ~20 mins
15 mins is in installing dependencies. caching doesn't seem to be working here. @fredyshox can you check that out?

- [ ] unit tests ~17 mins
There's probably a few unnecessarily slow tests here, fixed one of them in https://github.com/commaai/openpilot/pull/29262. in order to properly fix this, we need to make sure individual test case time is easily visible in the output and enforce a default timeout.

adeebshihadeh, 2023-08-17:
@fredyshox I unchecked macOS. it's under 10 mins, but half the time is spent on cleaning the scons cache, which should take almost no time.

fredyshox, 2023-08-17:
Working on that here: https://github.com/commaai/openpilot/pull/29401

jnewb1, 2023-08-21:
For car models:

[image]

About the time is panda_safety_carstate, which is 80% taken by the car interface update loop, which can't really be sped up.
I think our best bet here might just be parallelization with buildjet, we should be able to run these tests in parallel

adeebshihadeh, 2023-08-22:
@jnewb1 why are docs and simulator checked off? they took 9.5m and 10m on the last commit on master, while the build job took 4m https://github.com/commaai/openpilot/actions/runs/5940120032/job/16108007184

jnewb1, 2023-09-09:
below 10 minutes on average now, but we can still improve some stuff

closed 2023-09-09 by jnewb1