commaai/openpilot #29395: nav: multiline instructions cut off with lanes

sshane, 2023-08-14:
Couple hundred commits ago: [image]

Now: [image]

deanlee, 2023-08-18:
Not sure why, can't reproduce on my laptop

sshane, 2023-08-18:
I noticed it's not all instructions, only this one. Let me try to find the route

sshane, 2023-08-19:
So if there was a few more characters after `Freeway` it would add the correct spacing. But if it was just a few characters on the second line, the QWidget this layout was in wouldn't resize. https://github.com/commaai/openpilot/pull/29465 fixes this, as there's no QWidget wrapper any more.

closed 2023-08-19 by sshane