commaai/openpilot #29454: Multilang: check translation accuracy

sshane, 2023-08-17:
ie. with some API that can evaluate how accurate translations are. No translation should be egregiously wrong or inappropriate.

harshau007, 2023-08-20:
Hello @sshane 

Can I pick this issue and if possible, can you point me the initial file where to begin as I am new to this project?

adeebshihadeh, 2023-12-18:
Bounty is for a test we can run in CI on all commits. Requirements:
* check for any bad language
* (at least vaguely) checks for accuracy
* runs in CI on all commits
* can't add more than 30s of CI time

Be careful with using external APIs; it's unlikely we can merge something that relies on one, unless maybe it's free and doesn't require a token/auth.

adeebshihadeh, 2023-12-22:
Splitting this into two bounties since it seems we'll want two different solutions for the two problems:
- [ ] $100 bounty for the bad language check
- [ ] $200 for the accuracy check

PR #30783 (2023-12-31): multilang: bad language translation check
may fix #29454 (bad language check)

- [X] use [LDNOOBW](https://github.com/LDNOOBW/List-of-Dirty-Naughty-Obscene-and-Otherwise-Bad-Words) to check for bad language
- [X] handle incorrectly labeled bad words (override list ~~or local copy~~)
- [X] check numerusforms

```
comma@comma-33e0a0c6:/data/openpilot$ ./selfdrive/ui/tests/test_translations.py 
checking main_en
checking main_de
checking main_fr
Fchecking main_pt-BR
checking main_tr
checking main_ar
checking main_th
checking main_zh-CHT
checking main_zh-CHS
checking main_ko
checking main_ja
....s. [... cut]
closed 2023-12-31 by adeebshihadeh