commaai/openpilot #29458: more mypy coverage

sshane, 2023-08-18:


sshane, 2023-08-23:
`--local-partial-types` should be a good one to add, not too many failures, and the failures make sense:

```python
scripts/code_stats.py:16: error: Need type annotation for "imps" (hint: "imps: Set[<type>] = ...")  [var-annotated]
selfdrive/thermald/thermald.py:276: error: Right operand of "or" is never evaluated  [unreachable]
selfdrive/thermald/thermald.py:360: error: Statement is unreachable  [unreachable]
tools/sim/tests/test_carla_integration.py:22: error: Need type annotation for "processes" (hint: "processes: <type> | None = ...")  [var-annotated]
tools/sim/tests/test_carla_integration.py:23: error: Need type annotation for "carla_process" (hint: "carla_process: <type> | None = ...")  [var-annotated]
Found 5 errors in 3 files (checked 15 source files)

tools/sim/lib/manual_ctrl.py:16: error: Need type annotation for "axis_states" (hint: "axis_states: Dict[<type>, <type>] = ...")  [var-annotated]
tools/sim/lib/manual_ctrl.py:17: error: Need type annotation for "button_states" (hint: "button_states: Dict[<type>, <type>] = ...")  [var-annotated]
tools/sim/lib/manual_ctrl.py:91: error: Need type annotation for "axis_map" (hint: "axis_map: List[<type>] = ...")  [var-annotated]
tools/sim/lib/manual_ctrl.py:92: error: Need type annotation for "button_map" (hint: "button_map: List[<type>] = ...")  [var-annotated]
Found 4 errors in 1 file (checked 15 source files)

tools/gpstest/fuzzy_testing.py:25: error: Need type annotation for "con" (hint: "con: <type> | None = ...")  [var-annotated]
Found 1 error in 1 file (checked 15 source files)

tools/bodyteleop/web.py:24: error: Need type annotation for "pcs" (hint: "pcs: Set[<type>] = ...")  [var-annotated]
tools/bodyteleop/web.py:25: error: Need type annotation for "pm" (hint: "pm: <type> | None = ...")  [var-annotated]
tools/bodyteleop/web.py:25: error: Need type annotation for "sm" (hint: "sm: <type> | None = ...")  [var-annotated]
Found 3 errors in 1 file (checked 15 source files)
```

Nik-09, 2023-09-30:
Hi @sshane can I get assigned to this issue?
Thanks.

LoganxDev, 2023-11-22:
Adeeb mentioned on the last PR he didn't like the annotations. I was going to add in more flags and fix them up a module or a few at a time to make them manageable, but the new flags will cause a lot of annotation errors. Should this issue be closed if the stricter flags aren't wanted?

adeebshihadeh, 2023-12-12:
Closing as this is too ambiguous.

closed 2023-12-12 by adeebshihadeh