commaai/openpilot #29487: Subaru gen1 experimental longitudinal transmission fault on Crosstrek 2018

martinl, 2023-08-19:
### Describe the bug

I tested recently merged Subaru gen1 experimental long (https://github.com/commaai/openpilot/pull/28872) on Crosstrek 2018 and I am getting transmission fault on ignition, after safety mode is set. Route: `8de015561e1ea4a0|2023-08-19--16-50-12`

I managed to narrow down the issue by checking out https://github.com/commaai/openpilot/pull/28872/commits/80d634cf474521f0bd54ba34256e9382ab60f259 in locally cached feature-subaru-long branch. This fixed the transmission fault on ignition, openpilot engaged but eyesight faulted when braking. Route: `8de015561e1ea4a0|2023-08-19--21-58-10`

After reverting https://github.com/commaai/openpilot/pull/28872/commits/44f493fdf6a20ebedea320ee09044c4071a04810 eyesight did not fault when braking. Route: `8de015561e1ea4a0|2023-08-19--22-40-39`

### Which car does this affect?

Subaru Crosstrek 2018

### Provide a route where the issue occurs

8de015561e1ea4a0|2023-08-19--16-50-12

### openpilot version

012060ba328da69e7fd62531bd65e8d064b3f6c0

jnewb1, 2023-08-20:
[image]

ES_Status was blocked because inactive transmission rpm was set to 600 when it should be zero.

https://github.com/commaai/openpilot/pull/29496

sshane, 2023-08-21:
I think test_models might have caught this if we added a longitudinal route.

closed 2023-08-21 by jnewb1