commaai/openpilot #29494: Ford maximum cruise speed incorrect

magicparty, 2023-08-20:
### Describe the bug

my ford car shows the cruise speed in KPH, but when I change the openpilot display uint to metric the maximum cruise speed displayed on lefttop is wrong the current speed is correct only the maximum speed is wrong.

### Which car does this affect?

Ford Kuga 2022

### Provide a route where the issue occurs

32a35015aa4d5598|2023-08-20--18-22-54--21

### openpilot version

02e4e06

chiachunli08, 2023-08-21:
Same situation here with Focus MK4 in Taiwan.

incognitojam, 2023-08-22:
Is it off by 0-2 kph or completely incorrect?

chiachunli08, 2023-08-22:
When I set 50km/h on the car, it shows MAX 80 on the openpilot screen

incognitojam, 2023-08-22:
I've identified the problem, will need a little more work to implement a fix for you

PR #31463 (2024-03-21): Ford: handle metric cruise speed (v2)
**Description**

I found a signal which appears to match the IPC "Show km/h" setting.
Requires commaai/opendbc#1010. Previous attempt: https://github.com/commaai/openpilot/pull/29526.

Closes #29494.

**Verification**

- [x] Test in car and confirm that toggling the "Show km/h" setting does not result in the cruise speed shown in openpilot being incorrect.  Started in kph: `e886087f430e7fe7/2024-02-14--19-23-28`, in mph: `e886087f430e7fe7/2024-02-14--19-27-31`
- [x] Test in a non-English (metric) car: `32a35015aa4d5598/2024-03-03--14-58-48`
closed 2024-03-21 by sshane