commaai/openpilot #29539: c3's LTE & GPS lost signal at same time

TheOfficialJosh, 2023-08-22:
### Describe the bug

Both the GPS signal and LTE were lost around the same time, during segment 17. Started in a parking garage but had good GPS and LTE throughout drive until came around a bend under a small bridge. 30 seconds later, no LTE or GPS until the c3 was restarted after I was parked.

### Provide a route where the issue occurs

5130484aa8069bad|2023-08-22--17-29-46--17

### openpilot version

0.9.5 / master-ci / bc8dd48 / Aug 22

mvogttech, 2023-08-27:
I am experiencing a similar issue. However, on top of the loss of GPS signal and LTE service OP is unavailable with “Calibration Invalid: (Pitch: -5.5, Yaw: -1.8)”. 

Mount is in the same position since October, 2021. 

Maps UI also shows “waiting for GPS signal”. 

When GPS signal is available the calibration warning disappears and OP becomes available.

Avi0n, 2023-10-13:
Similar issue here. OP was working well until it lost GPS signal then it flashed a warning and disengaged. I can't remember if it also lost cell signal or not. OP would not re-engage until I re-started the car. I can't remember the exact message preventing OP from re-engaging, but I think it was "rawgpsd is not running".

20e394a46517a2ad|2023-10-07--14-46-32--95

adeebshihadeh, 2024-01-05:
Still having issues?

Avi0n, 2024-01-06:
No, I haven't seen this happen in a while. But I also haven't gone on any long road trips either since the issue occurred for me.

TheOfficialJosh, 2024-01-06:
I also haven't seen this issue since I reported it.

adeebshihadeh, 2024-01-06:
Cool, I’ll close and we can reopen if it pops up again.

closed 2024-01-06 by adeebshihadeh