commaai/openpilot #29552: IONIQ EV 2020 - LFA Warning on the dash when OP is running

dermida, 2023-08-23:
### Describe the bug

When OpenPilot is running the car shows on the dash a warning to "Check LFA system".

[image]

My car only has "LKA", but now thinks it has LFA and the warning appeared.

[image]

This warining started to appear now with the C3X, before with C2 didn't appear.

With the help of @sshane we were able to identify that: "When we used to send an LFA message to your car, which normally doesn't have LFA, it put the car into a persistent state where it thinks it has LFA, so our missing message is causing a fault"

Discord Thread: [car thinks it has LFA](https://discord.com/channels/469524606043160576/1143645816536436746)

### Which car does this affect?

HYUNDAI IONIQ ELECTRIC 2020

### Provide a route where the issue occurs

085e33b2085c44f4|2023-08-22--08-08-05

### openpilot version

0.9.5-lfa-msg-test (606cb6547c567582924c08df23bdc26e18a70524)

### Additional info

With the help of @sshane we tested a branch (lfa-msg-test (606cb6547c567582924c08df23bdc26e18a70524)) that forces sending de "LFA" message and with that branch all warnings disappeared

[image]

adeebshihadeh, 2023-09-11:
@sshane is this still an issue?

sshane, 2023-09-11:
Yes, the car (cluster, ABS?) is expecting the LFA message but is not receiving it because we detect the camera not sending it.

@dermida do the warning show up without the comma 3X plugged in? What about if the 3X is in dashcam mode? Can you get a route where it's in dashcam mode?

dermida, 2023-09-11:
> @dermida do the warning show up without the comma 3X plugged in? 
Yes. Even when not plugged in.

> What about if the 3X is in dashcam mode? 
Never tried. But i can try if you need.

> Can you get a route where it's in dashcam mode?
Sure. I will try tomorrow.

sshane, 2023-09-11:
Sounds like we put one of your ECUs into a state of expecting the LFA message before we detected it in https://github.com/commaai/openpilot/pull/27323. So, the current openpilot behavior is not a bug, and I don't think there's a solution to this that should be merged. I would recommend trying to visit your dealer to have them check it out and re-flash the ECU responsible.

The best we can do is try to narrow down the message that describes the fault, then we can have a good guess of which ECU we affected in the past. Feel free to reach out on Discord.

closed 2023-09-11 by sshane