commaai/openpilot #29627: Subaru: inconsitent torque across platforms

jnewb1, 2023-08-25:
### Describe the bug

We need to better unstand the torque differences across different subarus.

Here's a plot of 20,000 subaru segments:

[image]

Green shades are the Legacy / Outback
Red shades are various other gen1 cars

I'm not seeing a significant difference between the two cars, even with some people who have manually increased the limits or were running before the torque was decreased. We also have stock ES routes that go up to 1400 on extreme turns before it gives up:

[image]

Relevant (closed) PRs:
https://github.com/commaai/openpilot/pull/23530
https://github.com/commaai/openpilot/pull/29097

### Provide a route where the issue occurs

Subarus

### openpilot version

all

ClockeNessMnstr, 2023-08-25:
2017-2019? Impreza, 2018-2019 Crosstrek, 2020 Crosstrek hybrid are 3071 without errors. 
This was still less torque than the 2019 Forester which errors over 2047, but was fairly comparable at that point.

Emalton, 2023-09-12:
What can I do on my preglobal Subaru to help out as well?

AdamSLevy, 2023-10-11:
Any idea what the torque limit is for a 21 crosstrek?

closed 2024-06-14 by jnewb1