commaai/openpilot #29677: paramsd: vehicleModelInvalid after long drives

adeebshihadeh, 2023-08-28:
From 45937213b62ce5ed in support
```
45937213b62ce5ed|2023-08-26--07-44-16
45937213b62ce5ed|2023-08-24--08-03-02
45937213b62ce5ed|2023-08-23--08-00-57
45937213b62ce5ed|2023-08-22--08-05-04
45937213b62ce5ed|2023-08-21--08-02-24
45937213b62ce5ed|2023-08-20--08-15-22
45937213b62ce5ed|2023-08-19--08-19-51
45937213b62ce5ed|2023-08-18--08-02-28
45937213b62ce5ed|2023-08-17--08-00-06
45937213b62ce5ed|2023-08-15--08-21-52
45937213b62ce5ed|2023-08-14--07-57-12
45937213b62ce5ed|2023-08-12--08-14-18
45937213b62ce5ed|2023-08-11--08-02-47
45937213b62ce5ed|2023-08-10--08-00-59
45937213b62ce5ed|2023-08-09--08-04-54
45937213b62ce5ed|2023-08-08--07-57-56
45937213b62ce5ed|2023-08-07--08-14-44
45937213b62ce5ed|2023-08-05--07-50-39
45937213b62ce5ed|2023-08-04--08-04-37
45937213b62ce5ed|2023-08-03--08-20-42
45937213b62ce5ed|2023-08-02--07-54-27
45937213b62ce5ed|2023-07-31--07-30-08
45937213b62ce5ed|2023-07-29--06-45-39
45937213b62ce5ed|2023-07-28--08-07-36
45937213b62ce5ed|2023-07-27--09-47-52
45937213b62ce5ed|2023-07-17--08-16-02
45937213b62ce5ed|2023-07-15--08-17-28
45937213b62ce5ed|2023-07-14--08-16-10
45937213b62ce5ed|2023-07-13--08-25-20
45937213b62ce5ed|2023-07-12--08-01-20
45937213b62ce5ed|2023-07-11--08-06-15
45937213b62ce5ed|2023-07-10--07-53-15
45937213b62ce5ed|2023-07-07--12-55-43
45937213b62ce5ed|2023-07-07--11-47-14```

royjr, 2023-09-23:
Restarted car, popped up again, restarted c3 first then car, fixed. Steering became drunk. I see the recent pr, but I figured I'd report it. https://github.com/commaai/openpilot/pull/29831
```
1a5d045d2c531a6d|2023-09-22--16-05-47--12
1a5d045d2c531a6d|2023-09-22--16-05-47--13
```

[image]

adeebshihadeh, 2023-09-23:
@nuwandavek can you take a look?