commaai/openpilot #29750: C3 no internet connection after boot with 3rd party SIM card

andiradulescu, 2023-09-01:
### Describe the bug

Every time I start the C3, I have no internet (bars). The APN is set ("net"). I use a 3rd party SIM card (Orange Romania).

It was working fine with another SIM card with another carrier (also from Romania). It happened roughly after I changed SIM cards.

The only way I can make it work is going to APN and saving it again, without editing it. I need do this every time I plug my C3.

### Provide a route where the issue occurs

b387d7bf0f78af1b|2023-08-31--15-36-35

### openpilot version

0.9.4

### Additional info

Started happening on sunnypilot (twilsonco fork) and switched to official release3 for this bug report.

I preserved 3 consecutive routes:
[b387d7bf0f78af1b|2023-08-31--15-36-35](https://useradmin.comma.ai/?onebox=b387d7bf0f78af1b%7C2023-08-31--15-36-35)
[b387d7bf0f78af1b|2023-08-31--15-41-19](https://useradmin.comma.ai/?onebox=b387d7bf0f78af1b%7C2023-08-31--15-41-19)
[b387d7bf0f78af1b|2023-08-31--15-55-49](https://useradmin.comma.ai/?onebox=b387d7bf0f78af1b%7C2023-08-31--15-55-49)

adeebshihadeh, 2023-09-02:
Duplicate of #29205

closed 2023-09-02 by adeebshihadeh

andiradulescu, 2023-09-02:
Just to clarify for when #29205 is getting fixed. In my case the APN doesn’t dissapear from the menu. When I go to APN, the APN is already there.

Pressing Save in the APN menu, without editing it, restores the internet connectivity.

andiradulescu, 2023-09-02:
@adeebshihadeh how hard you think #29205 is to fix? do you think it could be a “good first issue”? I could try to fix it, if yes. thanks!

adeebshihadeh, 2023-09-02:
Yes it’s likely an easy fix.

andiradulescu, 2023-11-23:
https://github.com/commaai/openpilot/pull/30517