commaai/openpilot #29751: Problem when delpy ./start_openpilot_docker.sh in tools/sim

Dragoadri, 2023-09-01:
### Describe the bug

Im having a problem when deploying "sudo ./start_openpilot_docker.sh" I get the next error, saying tmux_script.sh is not found when yes, it is in the folder tools/sim: The error log: 
Just attend the last sentence 

What's Next?
  View summary of image vulnerabilities and recommendations → docker scout quickview ghcr.io/commaai/openpilot-sim:latest
Error response from daemon: Cannot kill container: openpilot_client: No such container: openpilot_client
**_/bin/bash: ./tmux_script.sh: No such file or directory_**

I must say **sudo ./start_carla.sh** works fine, and some weeks ago, everything worked fine, just tried again and get this issue.

### Provide a route where the issue occurs

none

### openpilot version

0.9.14

closed 2023-09-01 by jnewb1

lihuixian001, 2023-09-02:
It seems to be a bug, you need to make the following modifications in  start_openpilot_docker.sh :

OPENPILOT_ DIR="/tmp/openpilot"

Dragoadri, 2023-09-02:
THANK YOU SO MUCH! This solved it