commaai/openpilot #29758: EV6: can packet inconsistent on startup

sshane, 2023-09-02:
noticed while checking out a fingerprinting issue where the camera failed to respond. The can packet drops out without the multiplexing mode being changed.

[image]

route: ff2bd20623fcaeaa|2023-08-30--08-05-08

sshane, 2023-09-02:
@briskspirit can you check this out?

briskspirit, 2023-09-05:
That's when CAN2 core (bus1/OBD) started to reset because of TEC going up and no ack on the bus. 
[image]

sshane, 2023-09-06:
ah, so a side effect of the issue causing https://github.com/commaai/openpilot/issues/29689? Certain ECUs on bus 1 (non-OBD) failed to respond.

sshane, 2023-09-15:
@briskspirit are you saying that OBD issues caused an issue on bus 1? I can go swap out the comma power, I think it's been faulty for a while now.

briskspirit, 2023-09-15:
emm, maybe as a side effect, idk. but sometimes there is no ack for our transmitted messages.

adeebshihadeh, 2023-12-12:
Still an issue?

sshane, 2023-12-12:
Hmm, a few recent routes looked fine, but this one still looks like it has issues. Also missing all responses from OBD multiplexed queries (VIN and FW):

3f1a6480f940cf9a|2023-12-07--09-02-54

[image]

sshane, 2024-03-30:
Hasn't been an issue recently

closed 2024-03-30 by sshane