commaai/openpilot #29761: UI crashed and restarted during drive

TheOfficialJosh, 2023-09-02:
### Describe the bug

While driving, the UI crashed and restarted. openpilot did not disengage during this process, and the UI did restore after the crash.

### Provide a route where the issue occurs

5130484aa8069bad|2023-09-02--16-38-02--16

### openpilot version

0.9.5 / beeps / 94727c4 / Sep 01

adeebshihadeh, 2023-09-06:
@sshane can you check this out?

sshane, 2023-09-06:
When viewing the core dump with `gdb` after `apport-unpack`, it notifies me that the file is too short.

sshane, 2023-09-06:
```python
Program terminated with signal SIGABRT, Aborted.                                                                                                                                          
#0  0x0000007f83272d78 in ?? ()          
[Current thread is 1 (LWP 50616)]                                                            
(gdb) bt                                                                                     
#0  0x0000007f83272d78 in ?? ()                                                              
#1  0x0000007fed0a7948 in ?? ()                                                              
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

Possibly a duplicate of https://github.com/commaai/openpilot/issues/27328

sshane, 2023-09-13:
I don't see any crash logs, but see the crash in errorLogMessage. Can you let me know if this happens again?

sshane, 2023-09-13:
I don't see any crash logs, but see that it did crash in errorLogMessage. Do you see anything in `/data/tombstones/` or `/var/crash` for around this date and time?

TheOfficialJosh, 2023-09-13:
@sshane checking, it looks like /data/tombstones does not exist, and /car/crash is empty.

sshane, 2023-09-13:
You checked `/var/`, not /`car/`?

TheOfficialJosh, 2023-09-13:
That's is correct, I checked var/crash. Sorry, autocorrect.

[image]

deanlee, 2023-12-19:
I've encountered a few times where the uiState::updateState times out and gets kicked off by the watchdog, It is still unclear which part of the code caused it.

closed 2024-01-05 by adeebshihadeh