commaai/openpilot #29806: C3x won't connect to 2018 Nidec Civic

madcapclover, 2023-09-07:
### Describe the bug

After replacing working C2 with C3x, C3x would not connect to car and was displaying "Dashcam mode, Car Unrecognized". Attempt to fingerprint was made by following wiki and fW's from useradmin were already found under car.civic. 

Shane describes: 
"looks like canState1 (OBD port or radar bus) is going into a bus off state when fingerprinting" 

### Which car does this affect?

2018 Honda Civic Sedan - 4 Dr 

### Provide a route where the issue occurs

2cc806a8ed3a7df4|2023-08-24--07-13-38

### openpilot version

0.9.4

### Additional info

Thanks for all the help, I appreciate it!

sshane, 2023-09-07:
@briskspirit got no response from ECUs on bus 1 (OBD port)

[image]

briskspirit, 2023-09-07:
@sshane seems like you are checking not the route that is in issue post. 
By looking at the route in the issue I see problem with bus 2. Non-typical IRQ load and bunch of Form/Ack errors. 
Now if you checked another one and see an issue with OBD and no issues with bus2 - I would try to check all connections/cables. 

[image]

sshane, 2023-09-07:
Earlier routes seemed to fingerprint, make sure that one didn't fingerprint.

sshane, 2023-09-08:
@briskspirit user tried a new cable and reports it fingerprints again. No more errors in logs.

closed 2023-09-08 by sshane