commaai/openpilot #29810: long pressing main cruise enable stock longitudinal while openpilot long is enabled

Notstupe, 2023-09-07:
### Describe the bug

Hardware C3X
CANFD HKG HDA2 

Install master-ci or devel and enable experimental mode via toggle

[main cruise button](https://cdn.discordapp.com/attachments/734633317969952769/1149226910585462854/IMG_20230907_021531102.jpg)

Short pressing main cruise on enables openpilot as expected. 
Long pressing main cruise button enable stock longitudinal and can even override calibration mode locked out.

### Which car does this affect?

Ioniq 5 2023 limited (probably all CANDFD HKG cars)

### Provide a route where the issue occurs

11c1f1909ca37bca|2023-09-07--02-11-17--0, 11c1f1909ca37bca|2023-09-07--06-04-09--0

### openpilot version

master-ci

sshane, 2023-09-12:
- Handled the CAN version of this issue here: https://github.com/commaai/openpilot/issues/29810

sshane, 2023-09-12:
The two routes you sent you were already using stock longitudinal, so it's not overriding openpilot longitudinal as you claim.

Secondly, you were enabling a feature called [Manual Speed Limit Assist](https://github.com/commaai/opendbc/pull/950), which simply places a soft cap on the vehicle speed, unless you mash the accelerator pedal. It does not apply acceleration on its own, and openpilot remains disabled.

You can see you were applying acceleration the entire time:

<details>
<summary>View</summary>
[image]
</details>

And when you weren't pressing the accelerator, but MSLA was active, the vehicle was slowing down:

<details>
<summary>View</summary>
[image]
</details>

Similar situation in your first route, SCC and MSLA were never enabled together (we look at SCC to determine cruise status, which is correct):

<details>
<summary>View</summary>
[image]
</details>

sshane, 2023-09-12:
So this isn't an issue when using stock long (as in the routes you provided), as the stock SCC system refuses to enable. However with openpilot longitudinal, we try to enable in this state. I assumed that it was managed by the same ECU that manages SCC (camera/radar/ADAS unit) like on CAN, but something else actually does. It still is sent when the ADAS unit is disabled, and it comes from PT side on HDA 1 cars.

It never exists on ICE CAN FD cars, always exists on EV CAN FD cars, and only sometimes exists on hybrid CAN FD cars.

Will first block for EVs.

sshane, 2023-09-12:
thanks for the report!

closed 2023-09-12 by sshane