commaai/openpilot #29812: Vehicle parameter identification failed and car swerving

ncoop23, 2023-09-07:
### Describe the bug

Yesterday while driving on the freeway, comma popped up with vehicle parameter identification failed and started swerving wildly.  On looking at connect, it looks like there may have been GPS interference? Unknown why error showed up. 

### Which car does this affect?

2022 Non SC Bolt EUV Premire

### Provide a route where the issue occurs

0def4a390f6fe5c0|2023-09-06--20-35-36--16

### openpilot version

Masrer fd3b8a3d7511ad3d7c5aa0b82a1e2c28505e5a7f

### Additional info

Vehicle started swerving uncontrollably on the freeway.

ncoop23, 2023-09-07:
Comma is also coming up with GPS errors.

ncoop23, 2023-09-07:
[image]

haraschax, 2023-09-07:
https://github.com/commaai/openpilot/commit/55b391509d94c45000df6754f173ad446dc1b34c

Already fixed in master, yes it is GPS related.

closed 2023-09-07 by haraschax