commaai/openpilot #29822: macOS: change pypoetry configuration directory

royjr, 2023-09-07:
### Describe the bug

```
Configuration file exists at /Users/blah/Library/Preferences/pypoetry, reusing this directory.

Consider moving TOML configuration files to /Users/blah/Library/Application Support/pypoetry, as support for the legacy directory will be removed in an upcoming release.
```

### OS Version

Ventura 13.4.1

### openpilot version or commit

master

fredyshox, 2023-09-12:
Poetry changed the location of configuration files. It still recognises the old location if it exists, and tries to use it (https://github.com/python-poetry/poetry/blob/master/src/poetry/locations.py#L32). Looks like you've run the setup a while ago. This warning can be removed by removing `~/Library/Preferences/pypoetry` directory and then rerunning mac_setup.sh.

TLDR; 
```
rm -rf ~/Library/Preferences/pypoetry

# from openpilot dir
./tools/mac_setup.sh
```

closed 2023-09-12 by adeebshihadeh

royjr, 2023-09-12:
Looks good!