commaai/openpilot #29839: 2023 Hyundai Kona Electric Humming Sound when Override.

briantran33, 2023-09-08:
### Describe the bug

Whenever I try to override my 2023 Hyundai Kona Electric, there will be a humming sound. This sounds louder when the model steers to the right and I steer to the opposite side. Here are the routes to it. 

Route 1: d3a039910163c67a|2023-09-07--22-59-30--4 
https://cdn.discordapp.com/attachments/818670705843240961/1149585932832493649/IMG_4718.mov

Route 2: d3a039910163c67a|2023-09-07--23-04-27--0
https://cdn.discordapp.com/attachments/818670705843240961/1149586417169739857/IMG_4719.mov

### Which car does this affect?

Hyundai Kona 2023 Electric

### Provide a route where the issue occurs

d3a039910163c67a|2023-09-07--23-04-27--0

### openpilot version

https://github.com/commaai/openpilot/pull/26090

adeebshihadeh, 2023-09-08:
Only with longitudinal enabled from that PR?

briantran33, 2023-09-08:
Yes, OP long was enabled so that I can reproduce it at low speed.

adeebshihadeh, 2023-09-08:
Still not clear - is this a problem on openpilot release/master,i.e. does it happen with stock ACC?

briantran33, 2023-09-08:
Sorry for the confusion. This happened with me while I was trying Sunny's fork with MADS. I was told to reproduce it on master-ci or release but since my car stock set speed was too high at 20mph so I can't reproduce it. Sunny added the longitudinal so that I can set speed around 7mph so that I can reproduce that issue.

briantran33, 2023-09-08:
This happens in both stock SCC and openpilot longitudinal. I have been consistently replicating this behavior when for example: model wants to turn left and starts steering to the left, and I override steering and actively turns right.

sshane, 2023-09-13:
Does it happen only when steering past 90 degrees, or any resistance on the wheel causes this noise?

briantran33, 2023-09-13:
This is replicable by steering the opposite direction where openpilot is steering towards, it does not need to be past 90° to reproduce. The stronger I manually steer the opposite direction, the more pronounced the humming/grinding sound is.

sshane, 2023-09-15:
I see the EPS column output torque oscillate very heavily whenever you apply torque with openpilot enabled. However, driver torque cars' safety drops to zero when the user overrides past a certain point, and it starts oscillating before we drop the steer bit for the fault workaround, so I doubt it's that.

The fact that it oscillates like this when openpilot is commanding no torque is very concerning. I would try to make a branch to set the steer bit to 0 and enable and see if it still happens.

[image]

At certain points when overriding it appears to not oscillate so heavily, so does this only happen occasionally? And does this ever happen without openpilot engaged in the same scenarios? It could possibly be a bad driver torque sensor.

[image]

briantran33, 2023-09-15:
This can be replicated almost all the time when I override the steering opposite to openpilot's steering (openpilot steering 45 degrees, I override at -45 degrees, for example). Since this only happens when I override while openpilot is actively steering, this has not happened when openpilot is disengaged.

sshane, 2023-09-15:
Can you try to reproduce with stock LKAS (openpilot in dashcam mode)? It's also worth trying to reproduce disengaged, since I didn't see the oscillation every time you overrode.

briantran33, 2023-09-15:
I will do that when I get back.

briantran33, 2023-09-23:
@sshane 
d3a039910163c67a|2023-09-22--21-57-26--0 - This is stock LKAS when Openpilot is in dashcam mode. 
d3a039910163c67a|2023-09-22--22-04-00--0 - This is when disengaged.