commaai/openpilot #29845: CI: check PR target branch

adeebshihadeh, 2023-09-08:
https://github.com/Vankka/pr-target-branch-action

AbhinavGoel9, 2023-12-16:
@adeebshihadeh  I'm interested in contributing to this issue and noticed that the provided solution uses the GitHub CLI (`gh`). Before I proceed, I wanted to confirm if it's okay to use `gh` for closing pull requests. If there are any preferences or specific alternatives you'd like me to consider, please let me know.

jnewb1, 2023-12-16:
> @adeebshihadeh I'm interested in contributing to this issue and noticed that the provided solution uses the GitHub CLI (`gh`). Before I proceed, I wanted to confirm if it's okay to use `gh` for [...]

If it’s included in the Ubuntu 20.94 image then that’s fine

closed 2023-12-17 by adeebshihadeh