commaai/openpilot #29859: Ford: Escape 2020 fingerprinting issue

jnewb1, 2023-09-09:
### Describe the bug

bad response from engine

[20035.035255] fw_versions.py:309 - : iso-tp query bad response: (2016, None) - 0x7f2231
[20035.057011] fw_versions.py:309 - : iso-tp query bad response: (2016, None) - 0x621e1c0517
[20035.075456] fw_versions.py:309 - : iso-tp query bad response: (2016, None) - 0x621e3c0000
[20035.096026] fw_versions.py:309 - : iso-tp query bad response: (2016, None) - 0x62f43c0779
[20035.125009] fw_versions.py:309 - : iso-tp query bad response: (2016, None) - 0x62f43356
[20035.137130] fw_versions.py:309 - : iso-tp query bad response: (2016, None) - 0x7f2231
[20035.145368] fw_versions.py:309 - : iso-tp query bad response: (2016, None) - 0x410b3c
[20035.165932] fw_versions.py:309 - : iso-tp query bad response: (2016, None) - 0x41049d
[20035.184374] fw_versions.py:309 - : iso-tp query bad response: (2016, None) - 0x41057d
[20035.205114] fw_versions.py:309 - : iso-tp query bad response: (2016, None) - 0x410c0d85
[20035.274251] fw_versions.py:309 - : iso-tp query bad response: (2016, None) - 0x410d00

### Which car does this affect?

Ford Escape 2020

### Provide a route where the issue occurs

fc5df38c8fcae904|2023-09-08--19-26-41

### openpilot version

0.9.5

sshane, 2023-09-11:
Make sure to request logs for at least one route.

sshane, 2023-09-11:
This user's missing engine FW was added here: https://github.com/commaai/openpilot/pull/29372 which is not in release yet. Then around this time, they modified/re-flashed their engine with a custom tune which changes the reported FW. We have a PR that blocks these FW made here: https://github.com/commaai/openpilot/pull/29016. Similar case of this we blocked was here: https://github.com/commaai/openpilot/pull/28956

closed 2023-09-11 by sshane